arduino/ArduinoCore-arc32 #341: IOS ANCS: How to add solicitedServiceUUID?

danlsk, 2016-11-02:
@sandeepmistry has done a great job for the nrf BLEPeripheral library that includes remote service , characteristics and solicitedService UUID for IOS ANCS implementation. How can this be called from this Curie Peripheral library? 
1. Saving bonding data
2. Setup SolicitedServiceUUID

I suppose porting some of those remote service library would be good.

noelpaz, 2016-11-08:
We are working with @sandeepmistry  to implement a majority of his API to the CureiBLE library. We will inform you as soon if we have implemented this API. The head of the corelibs-arduino101 master branch does not have this implementation at this time and of course not in the official 1.0.7 official release that is downloadable from the Arduino IDE

noelpaz, 2016-11-16:
After reading more about this here https://developer.apple.com/library/content/documentation/CoreBluetooth/Reference/AppleNotificationCenterServiceSpecification/Introduction/Introduction.html, one of the requirements the Service Change Characterisitic is not going to be implemented soon as this requires low level firmware changes to the Nordic BLE stack.

I am closing this for now and maay reopen when the time comes or could be related to another pull request.  The CurieBLE is being refactored with close guidance from Arduino and will incrementally add features.

closed 2016-11-16 by noelpaz

sandeepmistry, 2016-11-16:
The biggest incremental feature that needs to be added is bonding/pairing support.

There are already plans to support solicited service UUID's and remote services and characteristics.

danlsk, 2017-04-07:
May I ask if there is any plan for this in 2017?

SidLeung, 2017-04-07:
There is no immediate plan to support the "bonding and pairing" feature in the CurieBLE library.

chartle2, 2017-04-11:
I would very much appreciate this feature as well. Hopefully it can be added to the plan in the not so distant future.