arduino/ArduinoCore-arc32 #347: BLE firmware version is not in sync with CurieBLE library

vigneshjathavara, 2016-11-22:
I replaced the ~/Library/Arduino15/packages/Intel/hardware/arc32/1.0.7/   directory with the code from Github as the readme said. Now whenever I try to upload a sketch, i get the following error:

!! BLE firmware version is not in sync with CurieBLE library !!
* Set Programmer to "Arduino/Genuino 101 Firmware Updater"
* Update it using "Burn Bootloader" menu
An error occurred while uploading the sketch

claudiaAl, 2016-11-22:
If I am not mistaken, this is because you also need to update your BLE firmware if you are wanting to use the latest code in the repository.  You can reference this closed issue [#322](https://github.com/01org/corelibs-arduino101/issues/322) and I think it should solve your problem.  Good luck!

kitsunami, 2016-11-28:
Hey, @eriknyquist Please add to the readme: update tools

eriknyquist, 2016-12-01:
Done. No response from @vigneshjathavara, closing.

closed 2016-12-01 by eriknyquist

davidbailey1018, 2016-12-28:
Starting download script...
Flashing is taking longer than expected
Try pressing MASTER_RESET button
!! BLE firmware version is not in sync with CurieBLE library !!
* Set Programmer to "Arduino/Genuino 101 Firmware Updater"
* Update it using "Burn Bootloader" menu
An error occurred while uploading the sketch.
i am using windows 7 and Arduino IDE 1.6.11.
how can i upload my code?

SidLeung, 2016-12-29:
Hello, There...
The Curie module has a special BLE processor that runs its own code, this is so called as the BLE firmware.  To fully utilize the potential of this processor, an updated of the BLE firmware is needed for this round of the release.
The downloading of the BLE f/w is done using the Arduino IDE.  The IDE will check for the right version of BLE f/w prior to running a sketch and will prompt the user for help to download the right version of f/w.
The statements logged above show the IDE detects the incorrect BLE version and it is providing the instructions for an user to update the f/w accordingly.
The user should find the "Programmer" selection under the "tool" tab and set it to "Arduino/Genuino 101 Firmware Updater".  Shown as follow,

[image]

After that, click the "Burn Bootloader" selection.  The BLE f/w download will follow.
Please note that, by doing the BLE f/w download this way, the user can jump back and forth between BLE firmware depending on the coreLibs release the user chooses to use.