arduino/ArduinoCore-arc32 #352: rssi measurement with ble in arduino genuino 101

doaaahme, 2016-12-01:
Hello
How can we measure the rssi of the peripheral board at the central board with ble ?

regards,
Doaa

noelpaz, 2016-12-01:
This is not released yet but this is a feature in the upcoming version. It is not in the master branch at the moment.

closed 2016-12-01 by eriknyquist

SidLeung, 2016-12-01:
Need to clarify here.  The current BLE library does NOT support the Central role.  In other words, a 101 board cannot be put into a Central role.

noelpaz, 2016-12-01:
@SidLeung -- This is not 1.0.7 Arduino release but what is currently in master and in the CurieBLE library we have the non release version of central which does not have the ability to query the rssi.

doaaahme, 2016-12-07:
Could we know when will that version be released?? the problem is we are working on a big project that needs this information before next January. would it be possible for this version to be released soon?