arduino/ArduinoCore-arc32 #353: disconnect from peripheral

doaaahme, 2016-12-01:
Hello

talking about IMUBleCentral & IMUBleNotification from CurieBle examples

why the peripheral board doesn't show that it has been disconnected from the central 
the peripheral doesn't appear to see those three commands
        digitalWrite(13, LOW);
        Serial.print("Disconnected from central: ");
        Serial.println(central.address());

it keeps the LED turned on even if we disconnect the central
regards,
Doaa

SidLeung, 2016-12-01:
Would you please check on the Central to see if it is stilled "paired" with the board?  If so, the Central has not released the Peripheral yet.

noelpaz, 2016-12-01:
hI @doaaahme  

Did you just power down the central or called end(0 or disconnect()?

noelpaz, 2016-12-14:
The reason for this is I think this is an callback context. Our suggestion is to limit code in callbacks and if possible just set certain states and have the loop do the work. Anyway disconnect and close is fixed in the next release.

closed 2016-12-14 by noelpaz