arduino/ArduinoCore-arc32 #354: Software serial does not work in 1.07

ilanganor, 2016-12-06:
software serial does not work properly in version 1.07

when connecting it to an actual device it returns "garbage"

the latest working version for the software serial is v1.04

reverting to 1.04 solved the issue

bigdinotech, 2016-12-09:
@ilanganor would you happen to have a scope capture of it failing?
Also what baud rate are you using when it fails. Or even better are you able to share your sketch that SoftwareSerial fails?

ilanganor, 2016-12-11:
Hello 
I do not have a scope capture of the signals.

I am working at 4800 baud rate and this sketch works great with version 1.04, however fails with versions 1.05, 1.06 and 1.07 

I also noticed I am not the only one with that problem:

[](https://forum.arduino.cc/index.php?topic=430581.0)

here is my test sketch:

[modem_At.txt](https://github.com/01org/corelibs-arduino101/files/644248/modem_At.txt)

paultanner, 2017-02-03:
Also failing when I use 1.04 (softwareSerial on pins 3 and 4).
Do these pins support softwareSerial?

kitsunami, 2017-02-24:
@bigdinotech , can you list the speeds that are supported?

bigdinotech, 2017-08-16:
fixed by
https://github.com/01org/corelibs-arduino101/commit/c794c47abf9864af34b768f46f0792991f12a649

closed 2017-08-16 by bigdinotech