arduino/ArduinoCore-arc32 #355: Runtime problem example

jimaobian, 2016-12-08:
Hi. When we try to use singleton, inherit and customized constructor at the same time this problem will occur. Please see the following example.
You can comment the customized constructor to eliminate this problem
```
class test1
{
  public:
    test1(){};  // You can comment this line to eliminate this runtime problem
    virtual void execute();  
};

class test2 :public test1
{
  public:
//    test2(){};  // You can comment this line to eliminate this runtime problem
    void execute(); 
    static test2* instance();
};

test2* test2::instance()
{
  static test2 state;
  return &state;
}

void test2::execute()
{
  Serial.println("hello");
}

test2 *mytest = NULL;

void setup() {
  Serial.begin(115200);
  while(!Serial);
  mytest = test2::instance();  
  Serial.println("test begin");
}

void loop() {
  mytest->execute();
  delay(100);
}
```

eriknyquist, 2016-12-09:
Hi @jimaobian, can we have a bit more information please? like, for example, 1) what is the expected outcome of your sketch above, and 2) what exactly is "the problem" (what goes wrong) ?

jimaobian, 2016-12-12:
Hi @eriknyquist 
Of course.
1. The outcome of the sketch I posted should print "hello" every 100ms. 
2. However, if I try to run it, it doesn't print anything.
3. If I comment the constructor of Class test1, it will print "hello" every 100ms.

facchinm, 2016-12-12:
Bug confirmed, the sketch works as expected on any other architecture (AVR, SAMD)

eriknyquist, 2016-12-12:
Thanks @facchinm and @jimaobian

eriknyquist, 2016-12-12:
Assigning to @calvinatintel and @SidLeung to investigate as toolchain issue

intel-larry, 2016-12-23:
I'm not sure this is really a good example of a singleton design pattern. I think the issue may be related to the newer C++ standards vs the older ones and the enforcement of publicly declared constructors. (have to think about what's going on here...)

eriknyquist, 2017-03-04:
Still trying to figure out what's wrong here-- I had hoped it was symptom of a bug that was fixed here https://github.com/01org/corelibs-arduino101/commit/d6c68f26914bf9508e8b76081d383a5cd432c7b6,
but unfortunately not. I should be able to work on this some more next week, I'll post any new information here