arduino/ArduinoCore-arc32 #366: CurieBLE: typed characteristic values not initialized to 0

sandeepmistry, 2016-12-29:
In v1.0.7 the following sketch would result in a characteristic with value `0`, however in master the characteristic has an empty value (value of length 0).

```arduino
#include <CurieBLE.h>

BLEPeripheral blePeripheral;  // BLE Peripheral Device (the board you're programming)
BLEService ledService("19B10000-E8F2-537E-4F6C-D104768A1214"); // BLE LED Service

BLEUnsignedCharCharacteristic switchCharacteristic("19B10001-E8F2-537E-4F6C-D104768A1214", BLERead | BLEWrite);

void setup() {
  Serial.begin(9600);

  blePeripheral.addAttribute(ledService);
  blePeripheral.addAttribute(switchCharacteristic);

  blePeripheral.begin();
}

void loop() {
  // listen for BLE peripherals to connect:
  BLECentral central = blePeripheral.central();

  // if a central is connected to peripheral:
  if (central) {
    Serial.print("Connected to central: ");
    // print the central's MAC address:
    Serial.println(central.address());

    // while the central is still connected to peripheral:
    while (central.connected()) {
      // ...
    }

    // when the central disconnects, print it out:
    Serial.print(F("Disconnected from central: "));
    Serial.println(central.address());
  }
}
```

SidLeung, 2017-01-07:
This issue is logged as Jira 804.

The issue suggests memory allocation error.

sgbihu, 2017-01-09:
This implementation is same as 1.0.7. I think it is because the value not be initialized. So the length is 0. I don't think this is a bug.

sandeepmistry, 2017-01-11:
I think the value is not set in the Nordic GATT table if it is `setValue` is called prior to adding the characteristic/service to `BLE`.

sgbihu, 2017-01-13:
So what result you want to see? From you test code, I don't see where you output the data.

BTW, the characteristic has two API, one is valueSize, another is valueLength. So if you don't set a value, the valueLength always 0.

sandeepmistry, 2017-01-24:
I expect a value of 0x00, if you look in [BLETypedCharacteristic.h](https://github.com/01org/corelibs-arduino101/blob/master/libraries/CurieBLE/src/BLETypedCharacteristic.h#L136-L143):

```
template<typename T> BLETypedCharacteristic<T>::BLETypedCharacteristic(const char* uuid, unsigned char properties) :
  BLECharacteristic(uuid, properties, sizeof(T))
{
    T value;
    memset(&value, 0x00, sizeof(value));

    setValue(value);
}
```

A `memset` + `setValue` is called in the constructor.

sgbihu, 2017-02-04:
All typed characteristic are called this constructor. That's why I don't think the code has bug.

sandeepmistry, 2017-02-07:
There's definitely an issue with setting characteristic values before adding them.

```arduino
#include <CurieBLE.h>

BLEPeripheral blePeripheral;

// create service
BLEService testervice = BLEService("FF10");

BLECharacteristic testCharacteristic = BLECharacteristic("FF11", BLERead, "hello");
void setup() {
  Serial.begin(9600);

  // set advertised local name and service UUID
  blePeripheral.setLocalName("test");
  blePeripheral.setDeviceName("test");
  blePeripheral.setAdvertisedServiceUuid(testervice.uuid());

  // add service and characteristics
  blePeripheral.addAttribute(testervice);
  blePeripheral.addAttribute(testCharacteristic);

  blePeripheral.begin();

  Serial.println(F("Bluetooth test"));
}

void loop() {
  blePeripheral.poll();
}
```

This sketch can also be used to reproduce, I expected the test characteristic to have a value of "hello" but it's empty when I read the value.

sgbihu, 2017-02-08:
I have duplicated this bug. And have solution now.

kitsunami, 2017-02-24:
Addressed by #433 .

sandeepmistry, 2017-03-15:
I'm still seeing the issue with the sample sketch from https://github.com/01org/corelibs-arduino101/issues/366#issue-198057880

Here's a screenshot from LightBlue on my Mac:

[image]

I would expect to see a value of `0x00` in LightBlue instead of the `0x` (which indicates value of 0 length).

sgbihu, 2017-03-16:
I think this is communication issue. The current implementation is not give the data until sketch set the value. So you mean the read response always get the characteristic's size.  Does below understand is correct?
char return 1 Byte
short return 2Bytes
string return the actual size that characteristic allocated. Like size is 16, string value is "Hello", the returned length is 16.

SidLeung, 2017-03-16:
@sandeepmistry , would like to clarify the expectation of the test result.  The Characteristic in the above sketch,

BLEUnsignedCharCharacteristic switchCharacteristic("19B10001-E8F2-537E-4F6C-D104768A1214"

had not been written to and was read.  Is there a reason to expect a length other than 0?  Understood the value of the characteristic should be 0 upon initialization but that should not be an indication of its availability.  Should it?

Please advise.

sandeepmistry, 2017-03-16:
@SidLeung this was the behaviour of the v1.0.7 release - all type characteristic type had values initialized to 0 (with now 0 length), for example:

 * byte, unsigned char, char: 0x00
 * short, unsigned short: 0x0000
 * int, unsigned int: 0x00000000

To re-iterate it's important for backwards compatibility. See #480 for a proposed fix.

#480 should also correct the behaviours when `characteristic.setValue(...)` or `characteristic.writeValue(...)` is called before the characteristic is added to a service.

SidLeung, 2017-03-16:
@sandeepmistry , risk of being repetitive here, upon the Characteristic initialization.  A Read of a characteristic will result in:

1. The value of the Characteristic is always 0.
2. The length of the Characteristic is the sizeof( CharacteristicType ).

sandeepmistry, 2017-03-16:
That is correct.

SidLeung, 2017-03-16:
Thank you and tracked by Jira 886.

SidLeung, 2017-03-17:
@sandeepmistry , please refer to the following conversation,

https://github.com/01org/corelibs-arduino101/issues/456#issuecomment-283950966

This user was not happy with the fact that the read() always return the sizeof (characteristicType) which, in his case, was 64 byte, a string array of char.

Isn't that is what #480?

sandeepmistry, 2017-03-20:
@SidLeung I would say https://github.com/01org/corelibs-arduino101/issues/456#issuecomment-283950966 is unrelated to this issue, as a typed characteristic is not used.

SidLeung, 2017-03-20:
Agreed.  Typed Characteristic was not used in this case.  However, should the return length be the size of the buffer defined by the Characteristic just like in the case of a typed characteristic?

sandeepmistry, 2017-03-20:
> However, should the return length be the size of the buffer defined by the Characteristic just like in the case of a typed characteristic?

The size should be fixed for typed characteristics only.

closed 2017-03-22 by SidLeung

sandeepmistry, 2017-03-24:
This looks good now with the 2.0.2 RC JSON @SidLeung provided me.

PR #480 (2017-03-22): CurieBLE: Set characteristic value internally if it not associated with a service yet
Resolves #366.

As outlined in the sketch provided in https://github.com/01org/corelibs-arduino101/issues/366#issue-198057880.

The initial value of 0x00 that was set by the type characteristic was dropped because it has not been associated with a service yet via `addCharacteristic(...)`. This change stores the value internally like the `BLECharacteristic(const char* uuid, unsigned char properties, const char* value)` constructor path.