arduino/ArduinoCore-arc32 #368: CurieBLE: BLE.scan() should scan with duplicate filter

sandeepmistry, 2016-12-29:
In https://github.com/01org/corelibs-arduino101/blob/master/libraries/CurieBLE/examples/central/scan/scan.ino

```arduino
BLE.scan();
```

scans without filter, however I expect it to scan with a duplicate filter.

SidLeung, 2017-02-24:
Track by Jira 802

kitsunami, 2017-02-24:
Addressed by #457

sandeepmistry, 2017-03-15:
This is a bit better with the 2.0.1 RC1 release, however the logic to enable the duplicate filtering is inverted.

I've submitted PR #476 to address this.

SidLeung, 2017-03-22:
Please note that PR #476 is superseded by the follow PR,

https://jira.ndg.intel.com/browse/ATLEDGE-896

sandeepmistry, 2017-03-24:
This looks good now with the 2.0.2 RC JSON @SidLeung provided me. Closing

The scan.ino example only reports discovered peripherals once as expected and the scan_callback.ino example reports every advertisement.

However, as @SidLeung mentioned `BLE.scan()` does not report the scan response as I expected. I'll open a separate issue to discuss this.

closed 2017-03-24 by sandeepmistry