arduino/ArduinoCore-arc32 #369: CurieBLE: BLE.scan(...) allows scans without duplicate filter

sandeepmistry, 2016-12-29:
In https://github.com/01org/corelibs-arduino101/blob/master/libraries/CurieBLE/examples/central/scan/scan.ino:

If I change the `BLE.scan();` line to either `BLE.scan(true);` or `BLE.scan(false);` all scan without duplicates. I expect it to only receive peripheral per Bluetooth address with the duplicate filter enabled.

SidLeung, 2017-01-06:
This issue is logged as Jira 802.

@sandeepmistry I think touch on this topic before.  The latest Nordic stack does not have readily available support for duplicate filtering.  Will investigating this issue further.

sandeepmistry, 2017-01-11:
Ok, if this is not supported by the Nordic soft device stack, the current API's are very confusing and we need to change them.

kitsunami, 2017-02-24:
Addressed by #457

sandeepmistry, 2017-03-15:
I've tested this with the 2.0.1 RC1 release, now a duplicate filter is present. Closing.

closed 2017-03-15 by sandeepmistry