arduino/ArduinoCore-arc32 #373: CurieBLE: Cannot discover all attributes on SensorTag

sandeepmistry, 2016-12-29:
I believe this one is in the JIRA system, however I'm opening it for other preview testers to track the issue.

I'm trying to use the [peripheral_explorer example](https://github.com/01org/corelibs-arduino101/blob/master/libraries/CurieBLE/examples/central/peripheral_explorer/peripheral_explorer.ino) with my TI SensorTag, however is fails to discover the attributes:

```
Found 68:C9:0BBLE Central - Peripheral Explorer
Found 68:C9:0B:06:BC:81 '' 
Connecting ...
Connected
Discovering attributes ...
Attribute discovery failed!

```

SidLeung, 2017-01-06:
This issue is logged as Jira 798.

Will be looking into increasing the Heap size to store all Attributes from scanning the SensorTag.

SidLeung, 2017-01-13:
The option of increasing ARC code SRAM size is being considered for the next release.  This is done with the replacement of the TD V1 f/w in the I86 with Zephyr RTOS v1.6

sandeepmistry, 2017-04-11:
This looks good now with the Bootes pre-release:

```
BLE Central - Peripheral Explorer
Found C8:69:CD:41:84:A2 '' 
Found 68:C9:0B:06:BC:81 'CC2650 SensorTag' aa80
Connecting ...
Connected
Discovering attributes ...
Attributes discovered

Device name: 
Service 1800
Service 1801
Service 180a
	Characteristic 2a23, properties 0x2, value 0x81BC0600000BC968
	Characteristic 2a24, properties 0x2, value 0x4343323635302053656E736F72546167
	Characteristic 2a25, properties 0x2, value 0x4E2E412E
	Characteristic 2a26, properties 0x2, value 0x312E323020284A756C203238203230313529
	Characteristic 2a27, properties 0x2, value 0x50434220312E322F312E33
	Characteristic 2a28, properties 0x2, value 0x4E2E412E
	Characteristic 2a29, properties 0x2, value 0x546578617320496E737472756D656E7473
	Characteristic 2a2a, properties 0x2, value 0xFE006578706572696D656E74616C
	Characteristic 2a50, properties 0x2, value 0x010D0000001001
Service f000aa00-0451-4000-b000-000000000000
	Characteristic f000aa01-0451-4000-b000-000000000000, properties 0x18, value 0x00000000
	Characteristic f000aa02-0451-4000-b000-000000000000, properties 0x10, value 0x00
	Characteristic f000aa03-0451-4000-b000-000000000000, properties 0x10, value 0x64
Service f000aa20-0451-4000-b000-000000000000
	Characteristic f000aa21-0451-4000-b000-000000000000, properties 0x18, value 0x00000000
	Characteristic f000aa22-0451-4000-b000-000000000000, properties 0x10, value 0x00
	Characteristic f000aa23-0451-4000-b000-000000000000, properties 0x10, value 0x64
Service f000aa40-0451-4000-b000-000000000000
	Characteristic f000aa41-0451-4000-b000-000000000000, properties 0x18, value 0x000000000000
	Characteristic f000aa42-0451-4000-b000-000000000000, properties 0x10, value 0x00
	Characteristic f000aa44-0451-4000-b000-000000000000, properties 0x10, value 0x64
Service f000aa80-0451-4000-b000-000000000000
	Characteristic f000aa81-0451-4000-b000-000000000000, properties 0x18, value 0x000000000000000000000000000000000000
	Characteristic f000aa82-0451-4000-b000-000000000000, properties 0x10, value 0x0002
	Characteristic f000aa83-0451-4000-b000-000000000000, properties 0x10, value 0x64
Service f000aa70-0451-4000-b000-000000000000
	Characteristic f000aa71-0451-4000-b000-000000000000, properties 0x18, value 0x0000
	Characteristic f000aa72-0451-4000-b000-000000000000, properties 0x10, value 0x00
	Characteristic f000aa73-0451-4000-b000-000000000000, properties 0x10, value 0x50
Service ffe0
	Characteristic ffe1, properties 0x16
		Descriptor 2901, value 0x4B6579205072657373205374617465
Service f000aa64-0451-4000-b000-000000000000
	Characteristic f000aa65-0451-4000-b000-000000000000, properties 0x10, value 0x7F
	Characteristic f000aa66-0451-4000-b000-000000000000, properties 0x10, value 0x00
Service f000ac00-0451-4000-b000-000000000000
	Characteristic f000ac01-0451-4000-b000-000000000000, properties 0x26, value 0x0000
	Characteristic f000ac02-0451-4000-b000-000000000000, properties 0x10, value 0x0202000000
	Characteristic f000ac03-0451-4000-b000-000000000000, properties 0x10, value 0x0044
Service f000ccc0-0451-4000-b000-000000000000
	Characteristic f000ccc1-0451-4000-b000-000000000000, properties 0x18, value 0x280000009001
	Characteristic f000ccc2-0451-4000-b000-000000000000, properties 0x8
	Characteristic f000ccc3-0451-4000-b000-000000000000, properties 0x8
Service f000ffc0-0451-4000-b000-000000000000
	Characteristic f000ffc1-0451-4000-b000-000000000000, properties 0x28
		Descriptor 2901,,value 0x496D67204964656E74696679
	Characteristic f000ffc2-0451-4000-b000-000000000000, properties 0x28
		Descriptor 2901, value 0x496D6720426C6F636B

Disconnecting ...
Disconnected
```

I ran the peripheral explorer example with a TI SensorTag.

SidLeung, 2017-04-18:
@sandeepmistry, Would you please close this issue if the code is satisfactory.

novashah, 2017-04-25:
Closing issue as resolved.

closed 2017-04-25 by novashah