arduino/ArduinoCore-arc32 #378: CurieBLE: cannot discover characteristics in 0x1800 and 0x1801 services

sandeepmistry, 2016-12-29:
I'm using the peripheral explorer with [bleno's echo example](https://github.com/sandeepmistry/bleno/tree/master/examples/echo), it doesn't report any characteristics in the 0x1800 and 0x1801 services.

Here's the output:

```
BLE Central - Peripheral Explorer
Found C8:69:CD:41:84:A2 '' 
Found 00:19:0E:11:24:06 '' ec00
Connecting ...
Connected
Discovering attributes ...
Attributes discovered

Device name: 
Service 1800
Service 1801
Service ec00
	Characteristic ec0e, properties 0x26

Disconnecting ...
Disconnected

```

SidLeung, 2017-01-06:
This issue is logged as Jira # &69.

@sandeepmistry Do you deem getting the default Services attributes necessary for the next release?

sandeepmistry, 2017-01-11:
@SidLeung I think we can live without it, but it would be nice to help with #379.

russmcinnis, 2017-05-02:
PR534 has resolved this issue using BLE V4.2,  not released yet. The jira is resolved.