arduino/ArduinoCore-arc32 #379: CurieBLE: peripheral.deviceName() returns empty String

sandeepmistry, 2016-12-29:
I'm using the peripheral explorer with [bleno's echo example](https://github.com/sandeepmistry/bleno/tree/master/examples/echo). `peripheral.deviceName()` return an empty string. I expect it to return `raspberrypi`.

Here's the output:
```
BLE Central - Peripheral Explorer
Found C8:69:CD:41:84:A2 '' 
Found 00:19:0E:11:24:06 '' ec00
Connecting ...
Connected
Discovering attributes ...
Attributes discovered

Device name: 
Service 1800
Service 1801
Service ec00
	Characteristic ec0e, properties 0x26

Disconnecting ...
Disconnected
```

Might be related to #378.

SidLeung, 2017-01-06:
This issue is logged as Jira 770.  This feature is currently not implemented.

@sandeepmistry Would you please provide the reasoning of using deviceName as oppose to localName of a Peripheral?

sandeepmistry, 2017-01-11:
Since the 0x2a00 characteristic is mandatory by the Core Bluetooth spec, it would be nice to have a convenience API to read the device name. It also allows you to store a longer name in the peripheral incase there is not enough space for the full desired local name to be advertised.

Once #378 is addressed this would be easier.