arduino/ArduinoCore-arc32 #382: CurieBLE: discovering attributes of connect central fails

sandeepmistry, 2016-12-30:
When running the following sketch, and connecting from LightBlue on my Mac, discovering attributes of the connected central fails:

```arduino
#include <CurieBLE.h>

void setup() {
  Serial.begin(9600);
  while(!Serial);

  BLE.begin();

  BLE.advertise();
}

void loop() {
  BLEDevice central = BLE.central();

  if (central) {
    Serial.print("Central connected: ");
    Serial.println(central.address());

    if (central.discoverAttributes()) {
      Serial.println("Discovered attributes");
    } else {
      Serial.println("Discovering attributes failed!");
    }

    while(central.connected()) {
      
    }

    Serial.print("Central disconnected: ");
    Serial.println(central.address());
  }
}
```

noelpaz, 2017-01-02:
I thought that discoverAttributes works only in central mode and not meant for the peripheral to discover central. I am also adding @sgbihu to this and hope he answers. I am still on vacation.

noelpaz, 2017-01-02:
It looks like I cannot assign @sgbihu . @eriknyquist can you send him an email to look into this

sandeepmistry, 2017-01-03:
> I thought that discoverAttributes works only in central mode and not meant for the peripheral to discover central. 

@noelpaz as discussed with @SidLeung during the initial calls, both centrals and peripherals can operate a GATT server and GATT client.

sgbihu, 2017-01-04:
This is a bug. The program doesn't add the connected central to the profile buffer. And that makes discover failed. The discover can't find it in the buffer.

sgbihu, 2017-01-04:
Hi @sandeepmistry ,

You can get the fix from https://github.com/sgbihu/corelibs-arduino101. The commit is a8c834a.

Thanks!

SidLeung, 2017-01-05:
Please note Jira ticket, 788, has been created.  A solution has been created for internal review, approval, and testing.  Upon the completion of that process, the solution will be merged to the master, will be in the nightly JSON build, and will post the status here.

noelpaz, 2017-01-25:
Tested and works accordingly.

sandeepmistry, 2017-01-26:
@noelpaz was the fix for this merged into master?

kitsunami, 2017-02-24:
This was addressed in #410

SidLeung, 2017-03-01:
PR #410 addresses this issue.

sandeepmistry, 2017-03-15:
I'm still seeing this issue with the sketch from https://github.com/01org/corelibs-arduino101/issues/382#issue-198154723 and my Mac running macOS 10.12.3.

The Serial monitor displays the following when I connect from LightBlue on my Mac:

```
Central connected: AC:BC:32:7C:FA:26
Discovering attributes failed!
Central disconnected: 
```

Here's a screenshot from the PacketLogger app, I don't see any incoming attribute discovery requests either:

[image]

sgbihu, 2017-03-16:
Hi @sandeepmistry ,
I have test those two sketches https://github.com/01org/corelibs-arduino101/tree/master/libraries/CurieBLE/examples/test/discoveratperipheral and https://github.com/01org/corelibs-arduino101/tree/master/libraries/CurieBLE/examples/test/profileatcentral. They doesn't have problem.

So I want to confirm your does your light blue has profiles? Please tell me the setups if you have. And does iPhone's light blue has same results? I don't have MAC OS to duplicate this issue. It's better we can do on IOS.

Thanks!

sandeepmistry, 2017-03-16:
I tried this again today with Github master and the 2.0.1 RC JSON. The sketch I provided works with both macOS and iOS.

Not exactly sure why it was failing yesterday, it could be that the GATT server on the Mac had too many attributes like https://github.com/01org/corelibs-arduino101/issues/479#issuecomment-287106847.

I'll close this for now, and re-open if I see it again.

closed 2017-03-16 by sandeepmistry