arduino/ArduinoCore-arc32 #383: CurieBLE: characteristic.read() does not block

sandeepmistry, 2016-12-30:
When trying the peripheral explorer example, I removed the `delay(1000);` statement inside the `exploreCharacteristic` as it was not in the original sample sketch we provided. When this was done the characteristic values were not printed to the Serial Monitor:

Before:

```
BLE Central - Peripheral Explorer
Found C8:69:CD:41:84:A2 '' 
Found E4:1E:14:D2:31:E7 '' 
Found AC:BC:32:7C:FA:26 'test1' ffffffff-ffff-ffff-ffff-fffffffffff0
Connecting ...
Connected
Discovering attributes ...
Attributes discovered

Device name: 
Service 1800
Service 1801
Service 180a
	Characteristic 2a29, properties 0x2, value 0x4170706C6520496E63
	Characteristic 2a24, properties 0x2, value 0x4D6163426F6F6B50726F31322C31
Service d0611e78-bbb4-4591-a5f8-487910ae4366
	Characteristic 0000, properties 0x24
Service ffffffff-ffff-ffff-ffff-fffffffffff0
	Characteristic 0000, properties 0x2, value 0x76616C7565
	Characteristic 0000, properties 0x2, value 0x64796E616D69632076616C7565
	Characteristic 0000, properties 0x2, value 0x000102030405060708090A0B0C0D0E0F10111213
	Characteristic 0000, properties 0x12
	Characteristic 0000, properties 0x16
	Characteristic 0000, properties 0x32

Disconnecting ...
Disconnected

```

After removing `delay`:

```
BLE Central - Peripheral Explorer
Found E4:1E:14:D2:31:E7 '' 
Found C8:69:CD:41:84:A2 '' 
Found C8:69:CD:41:84:A2 '' 
Found AC:BC:32:7C:FA:26 'test1' ffffffff-ffff-ffff-ffff-fffffffffff0
Connecting ...
Connected
Discovering attributes ...
Attributes discovered

Device name: 
Service 1800
Service 1801
Service 180a
	Characteristic 2a29, properties 0x2
	Characteristic 2a24, properties 0x2
Service d0611e78-bbb4-4591-a5f8-487910ae4366
	Characteristic 0000, properties 0x24
Service ffffffff-ffff-ffff-ffff-fffffffffff0
	Characteristic 0000, properties 0x2
	Characteristic 0000, properties 0x2
	Characteristic 0000, properties 0x2
	Characteristic 0000, properties 0x12
	Characteristic 0000, properties 0x16
	Characteristic 0000, properties 0x32

Disconnecting ...
Disconnected
```

Once this is fixed, the  `delay(1000);` statement should be removed from the sketch.

SidLeung, 2017-01-06:
The previous release, v1.0.7, BLE implementation employed end-to-end command/reply handshaking for each operation (read, write, etc).  Thus, the overall BLE performance was substantially impacted.  The latest BLE library uses the request/query approach that enables non-blocking operation to improve performance and bandwidth utilization.

To be backward compatible, may need to introduce a bool flag, as an input parameter, for blocking call selection.  By default, an operation will be blocking.

This request for change is logged as Jira 795.

sandeepmistry, 2017-01-11:
@SidLeung 

> To be backward compatible, may need to introduce a bool flag, as an input parameter, for blocking call selection. By default, an operation will be blocking.

I'm confused, there was no `read` operation in v1.0.7. So, I expect it to always block.

sgbihu, 2017-01-12:
The read is for central to read attributes.

SidLeung, 2017-02-08:
Resolution passed code review, system testing, and merged to main trunk.  Will be in the next nightly JSON build.

sandeepmistry, 2017-03-15:
Testing with the 2.0.1 RC1 release, this is now fixed. Closing.

```
BLE Central - Peripheral Explorer
Found 69:0D:D0:1D:01:B8 '' 
Found 69:0D:D0:1D:01:B8 'PizzaSquat' 
Connecting ...
Connected
Discovering attributes ...
Attributes discovered

Device name: 
Service 1800
Service 1801
Service 180a
	Characteristic 2a29, properties 0x2, value 0x4170706C6520496E63
	Characteristic 2a24, properties 0x2, value 0x4D6163426F6F6B50726F31322C31
Service d0611e78-bbb4-4591-a5f8-487910ae4366
	Characteristic 8667556c-9a37-4c91-84ed-54ee27d90049, properties 0x24
Service 9fa480e0-4967-4542-9390-d343dc5d04ae
	Characteristic af0badb1-5b99-43cd-917a-a77bc549e3cc, properties 0x24
Service 13333333-3333-3333-3333-333333333337
	Characteristic 13333333-3333-3333-3333-333333330001, properties 0x10, value 0x00
		Descriptor 2901, value 0x
	Characteristic 13333333-3333-3333-3333-333333330002, properties 0x10
		Descriptor 2901, value 0x
	Characteristic 13333333-3333-3333-3333-333333330003, properties 0x24
		Descriptor 2901, value 0x

Disconnecting ...
Disconnected
```

closed 2017-03-15 by sandeepmistry