arduino/ArduinoCore-arc32 #384: CurieBLE: Characteristic UUID's have incorrect value of 0000

sandeepmistry, 2016-12-30:
When using the [pizza example from bleno](https://github.com/sandeepmistry/bleno/tree/master/examples/pizza) on my Mac when the peripheral explorer CurieBLE example, a few characteristics have UUID values of `0000` which is incorrect. Here's the output from the Serial Monitor:

```
BLE Central - Peripheral Explorer
Found C8:69:CD:41:84:A2 '' 
Found E4:1E:14:D2:31:E7 '' 
Found AC:BC:32:7C:FA:26 '' 13333333-3333-3333-3333-333333333337
Connecting ...
Connected
Discovering attributes ...
Attributes discovered

Device name: 
Service 1800
Service 1801
Service 180a
	Characteristic 2a29, properties 0x2, value 0x4170706C6520496E63
	Characteristic 2a24, properties 0x2, value 0x4D6163426F6F6B50726F31322C31
Service d0611e78-bbb4-4591-a5f8-487910ae4366
	Characteristic 0000, properties 0x24
Service 13333333-3333-3333-3333-333333333337
	Characteristic 0000, properties 0x10, value 0x00
	Characteristic 0000, properties 0x10, value 0x0000
	Characteristic 0000, properties 0x24

Disconnecting ...
Disconnected
```

If I connect to the Mac peripheral using LightBlue on iOS the characteristic UUID's value match what I expect.

sgbihu, 2017-01-09:
Can you describe your test instructions? I suspect the characteristic's UUID are not sub-UUID with services. And that makes the stack return an error. The characteristic's UUID in the below bold text is different with service UUID, then the discover can success. If the other is not same. I think it may have the error you described.

19b1**0000**-e8f2-537e-4f6c-d104768a1214

sandeepmistry, 2017-01-11:
I think this is specific to bleno on OS X, as OS X has some "secure" characteristics used for hand off with iOS devices.

Test instructions:
1. Obtain a Bluetooth 4.0 capable Mac (I'm using 10.11.x)
2. Install Node.js
3. Make a test directory
4. Run `npm install bleno`
5. `cd node_modules/bleno`
6. Run `node examples/pizza/peripheral.js`

sgbihu, 2017-01-13:
Hi @sandeepmistry ,

The 0000 UUID is expected after I review your test code. Because the UUID not satisfy the Nordic's stack. It will return unknown UUID and this UUID is 0.
uuid: '13333333-3333-3333-3333-333333333337'
below is characteristics' UUID.
uuid: '13333333-3333-3333-3333-333333330001'
uuid: '13333333-3333-3333-3333-333333330002'
uuid: '13333333-3333-3333-3333-333333330003'

If you change the characteristic's UUID as below. It will works.

uuid: '13333334-3333-3333-3333-333333333337'
uuid: '13333335-3333-3333-3333-333333333337'
uuid: '13333336-3333-3333-3333-333333333337'

sandeepmistry, 2017-01-24:
What makes them invalid UUIDs? Other central apps/stacks like LightBlue, nRF Master control panel have no issues with this example.

SidLeung, 2017-01-27:
Sandeep:

Is this feature needed for the next official release?

sandeepmistry, 2017-01-27:
Yes please.

From my perspective this is an issue/bug not a feature.

SidLeung, 2017-03-01:
Jira ticket 793 with PR #451 are created for addressing this issue.

sandeepmistry, 2017-03-15:
Testing with the 2.0.1 RC1 release, this looks good now. Closing.

```
BLE Central - Peripheral Explorer
Found 17:16:01:7B:A4:FB '' 
Found 69:0D:D0:1D:01:B8 '' 
Found 69:0D:D0:1D:01:B8 'PizzaSquat' 
Connecting ...
Connected
Discovering attributes ...
Attributes discovered

Device name: 
Service 1800
Service 1801
Service 180a
	Characteristic 2a29, properties 0x2, value 0x4170706C6520496E63
	Characteristic 2a24, properties 0x2, value 0x4D6163426F6F6B50726F31322C31
Service d0611e78-bbb4-4591-a5f8-487910ae4366
	Characteristic 8667556c-9a37-4c91-84ed-54ee27d90049, properties 0x24
Service 9fa480e0-4967-4542-9390-d343dc5d04ae
	Characteristic af0badb1-5b99-43cd-917a-a77bc549e3cc, properties 0x24
Service 13333333-3333-3333-3333-333333333337
	Characteristic 13333333-3333-3333-3333-333333330001, properties 0x10, value 0x00
	Characteristic 13333333-3333-3333-3333-333333330002, properties 0x10, value 0x0000
	Characteristic 13333333-3333-3333-3333-333333330003, properties 0x24

Disconnecting ...
Disconnected
```

closed 2017-03-15 by sandeepmistry