arduino/ArduinoCore-arc32 #387: ERROR Compile the StandardFirmataBle

hulkco, 2016-12-30:
I have this error when updating the latest snapshot:
In file included from /tmp/arduino_build_563639/sketch/bleConfig.h:65:0,
                 from /root/Arduino/libraries/Firmata/examples/StandardFirmataBLE/StandardFirmataBLE.ino:42:
/root/Arduino/libraries/Firmata/utility/BLEStream.h: In member function ‘void BLEStream::end()’:
/root/Arduino/libraries/Firmata/utility/BLEStream.h:121:59: error: call of overloaded ‘setEventHandler(BLECharacteristicEvent, NULL)’ is ambiguous
   this->_rxCharacteristic.setEventHandler(BLEWritten, NULL);
                                                           ^
/root/Arduino/libraries/Firmata/utility/BLEStream.h:121:59: note: candidates are:
In file included from /root/.arduino15/packages/Intel/hardware/arc32/1.0.7/libraries/CurieBLE/src/CurieBLE.h:37:0,
                 from /tmp/arduino_build_563639/sketch/bleConfig.h:64,
                 from /root/Arduino/libraries/Firmata/examples/StandardFirmataBLE/StandardFirmataBLE.ino:42:
/root/.arduino15/packages/Intel/hardware/arc32/1.0.7/libraries/CurieBLE/src/BLECharacteristic.h:465:10: note: void BLECharacteristic::setEventHandler(BLECharacteristicEvent, BLECharacteristicEventHandler)
     void setEventHandler(BLECharacteristicEvent event, 
          ^
/root/.arduino15/packages/Intel/hardware/arc32/1.0.7/libraries/CurieBLE/src/BLECharacteristic.h:467:10: note: void BLECharacteristic::setEventHandler(BLECharacteristicEvent, BLECharacteristicEventHandlerOld)
     void setEventHandler(BLECharacteristicEvent event,

hulkco, 2016-12-30:
Sorry i have need Firmata ble under Test Folder

sandeepmistry, 2016-12-30:
I think this issue should be re-opened until the following pull request is merged by the Firmata maintainers: https://github.com/firmata/arduino/pull/335

hulkco, 2016-12-30:
I reopen this issue for a request by @sandeepmistry

eriknyquist, 2016-12-30:
Sandeep already filed an issue for this, so I don't know why he asked you to re-open it.

https://github.com/01org/corelibs-arduino101/issues/377

Please follow that one ^^

don't re-open this one.

closed 2016-12-30 by eriknyquist

sandeepmistry, 2016-12-30:
@eriknyquist #377 is about the sketch not working after https://github.com/firmata/arduino/pull/335 is applied.