arduino/ArduinoCore-arc32 #388: ERROR: I can not upload any sketches.

hulkco, 2016-12-30:
ERROR: I can not upload any sketches.

java.io.IOException: Cannot run program "/root/.arduino15/packages/Intel/tools/arduino101load/1.6.9+1.28/arduino101load/arduino101load": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:207)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:709)
	at processing.app.SketchController.exportApplet(SketchController.java:682)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2189)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 10 more
An error occurred while uploading the sketch

hulkco, 2016-12-31:
Yes i update the tools to

hulkco, 2017-01-01:
Hello everyone, I already found the solution, without realizing it, download the tools for another platform, so I uploaded suggestions to the README.md file so that the instructions are clearer.
Greetings.

closed 2017-01-01 by hulkco

eriknyquist, 2017-01-04:
Hi @hulkco, I don't see a pull request for the README change you mentioned, but we'd be happy to make things clearer in there if you have some suggestions :)

hulkco, 2017-01-06:
Hello, I'm new to GitHub, as I uploaded the suggested changes to README, and create the README for intel-arduino-tools, doing so automatically became a FORK, and I thought it would only wait for the supervisor to accept the changes, I do not know , what has happened.

eriknyquist, 2017-01-06:
@hulkco if you want to email me a patch, I'd be happy to commit the changes for you this time. For future changes, here is good tutorial on creating Pull Requests https://yangsu.github.io/pull-request-tutorial

hulkco, 2017-01-06:
Thanks I will try to make a summary and I will send it to you as soon as possible, in fact, I did a small tutorial in hackster.io [Upgrading the firmware & library of Arduino 101](https://www.hackster.io/Hulkco/upgrading-the-firmware-library-of-arduino-101-4d6b10)