arduino/ArduinoCore-arc32 #390: CurieTimer1Interrupt example sketch bugs

packetmax, 2017-01-04:
I just got round to getting a couple of 101's going and noticed that the example sketch CurieTimer1Interrupt uses type int to express microsecond multiples. Change both vars oneSecInUsec and time (bad naming choice) to long int and you're good to go.

eriknyquist, 2017-01-04:
Looks fine to me.... the function expectes an int

    inline int start(const unsigned int timerPeriodUsec, void (*userCallBack)() = NULL)

and it looks like `time` will always be less than 1000000. Am I missing something?

packetmax, 2017-01-04:
Int vars are 2 bytes (1000000 needs 12 bits :). I didn't dig thru the
library headers, but there likely is a version that accepts a long. The
sketch didn't work until I made said changes.

packetmax, 2017-01-04:
Make that 12 a 20 :).

packetmax, 2017-01-04:
It's my mistake Erik. I did a println(sizeof(int)) on the 101 and the
answer is 4. I'm not sure what was happening to cause what I saw.
Anyway--you're correct; I am not. Thanks, -g

eriknyquist, 2017-01-04:
No worries :) Glad you got it working.

FYI, the C standard specifies that an `int` must be *at least* 2 bytes, but on most implementations (including in the Arduino101), it is 4. You can prove it with this sketch:

    int num = 2147483647;

    void setup() {
      Serial.begin(9600);
    }

    void loop() {
      delay(2000);
      Serial.println(num);
    }

eriknyquist, 2017-01-04:
Or you can check with `sizeof` like you did! :)

eriknyquist, 2017-01-04:
@packetmax your concerns are not unfounded, though. It's not very portable to use an `int` like we are here-- I might change it to `long` anyway :)

packetmax, 2017-01-04:
> int num = 2147483647 <(214)%20748-3647>;

Yeah, I should know better Erik. cheers! -g

packetmax, 2017-01-04:
Personally I'm a fan of [u]int16_t, [u]int32_t, etc for portability and
just understanding the teleonomic aspects of someone else's code :).
Sometimes explication beats elegance :).

eriknyquist, 2017-01-04:
As am I. Although the Arduino Community generally prefers the less-scary looking types like `int` and `long int` (even though you'll notice we're not terribly good at conforming to that in all of our libraries!)

closed 2017-01-04 by eriknyquist