arduino/ArduinoCore-arc32 #392: Can I use CurieTimerOne inside of another library?

claudiaAl, 2017-01-06:
Hello!  I am having issues converting my sketch to a library.  Specifically, I am trying to start CurieTimerOne from my library.  In an effort to pinpoint the problem, I took the CurieTimer1Interrupt example and modified it to fit what I am trying to accomplish.  From my example sketch, I am creating an instance of my class, and calling the Initialize() function which in turn calls CurieTimerOne.start(freq, ISR).  Everytime the ISR is called, the led should be toggled.  It looks like as soon as the sketch is loaded, the led gets turned on, but never toggles.  I thought initially that it might be because an instance of CurieTimerOne is automatically created when a sketch is started, but if the library is only included once in my header file, I'm not really sure that that's the issue.  

I'm attaching a copy of my lib folder to run the simple example.  Can anyone educate me on where the issue lies?

Thanks!!

.h
```
#ifndef testLib_Timer1_h
#define testLib_Timer1_h

#include <CurieTimerOne.h>

class testLib_Timer1
{
public:
    void Initialize();
    void toggleLed(); 
protected:
};
#endif /* testLib_Timer1_h */
```
.cpp
```
#include "testLib_Timer1.h"
#include "CurieTimerOne.h"

const int oneSecInUsec = 1000000;   // A second in mirco second unit.
bool toggle = 1;

void timedBlinkISR()
{
    digitalWrite(13,toggle);
    toggle = !toggle;
}

void testLib_Timer1::Initialize()
{
    pinMode(13, OUTPUT);
    CurieTimerOne.start(1000, timedBlinkISR);
}

void testLib_Timer1::toggleLed()
{
    if(toggle == 0)
    {
        digitalWrite(13,HIGH);
        toggle = 1;
    }
    else
    {
        digitalWrite(13, LOW);
        toggle = 0;
    }
}
```
.ino
```
#include <testLib_Timer1.h>

testLib_Timer1 myTest;

void setup() {
  myTest.Initialize();
}

void loop() {
}
```

[testLib_Timer1.zip](https://github.com/01org/corelibs-arduino101/files/690968/testLib_Timer1.zip)

eriknyquist, 2017-01-07:
Hi @claudiaAl, we try to keep only defects and feature requests in the Github Issue tracker.
Would you mind re-posting this on arduino.cc ?

https://forum.arduino.cc/index.php?board=103.0

(I hang out there too :))

claudiaAl, 2017-01-07:
Yes, no problem, thank you!

http://forum.arduino.cc/index.php?topic=446520.0

closed 2017-01-07 by claudiaAl

eriknyquist, 2017-01-07:
OK. For future posts, there is actually a specific Arduino101 section, which the URL in my previous post will take you to :)

claudiaAl, 2017-01-07:
I had actually posted it earlier today, but I didn't know there was an Arduino101 section!  Thank you! I'll go ahead and post my link there as well.

eriknyquist, 2017-01-07:
No need to cross-post, just wanted to let you know for future posts :)