arduino/ArduinoCore-arc32 #398: can't measure the rssi while being connected

doaaahme, 2017-01-12:
*/
#include <CurieBLE.h>

/*
  This sketch example works with IMUBleNotification.ino

  IMUBleNotification.ino will send notification to this central sketch.
  This sketch will receive the notifications and output the received data in the serial monitor.
  It also illustrates using a non-typed characteristic.
  Set the baud rate to 115200 on the serial monitor to accomodate the speed of constant data updates from IMU subsystem.
*/

#define LED_PIN   13
#define MAX_IMU_RECORD 1

// define a structure that will serve as buffer for holding IMU data

typedef struct {
    int index;
    unsigned int slot[3];
} imuFrameType;

imuFrameType imuBuf[MAX_IMU_RECORD];

void setup()
{
    // This is set to higher baud rate because accelerometer data changes very quickly
    Serial.begin(9600);    // initialize serial communication
    while (!Serial);
    pinMode(LED_PIN, OUTPUT);    // initialize the LED on pin 13 to indicate when a central is connected
    
    /* Now activate the BLE device.  It will start continuously transmitting BLE
     advertising packets and will be visible to remote BLE central devices
     until it receives a new connection */
    BLE.begin();
    Serial.println(BLE.address());
    
    BLE.scanForName("LED");
}

void loop()
{
    BLEDevice peripheral = BLE.available();
    //pr_debug(LOG_MODULE_BLE, "%s-%d",__FUNCTION__, __LINE__);
    if (peripheral) 
    {
        Serial.println(peripheral.address());
        BLE.stopScan();

       
        delay (1000);
        // central connected to peripheral
        controlImu(peripheral);
        delay (4000);
        BLE.scanForName("LED");
    }
}

void controlImu(BLEDevice peripheral)
{
    static bool discovered = false;
    // connect to the peripheral
    Serial.print("Connecting ... ");
    Serial.println(peripheral.address());

    if (peripheral.connect())
    {
        Serial.print("Connected: ");
        Serial.println(peripheral.address());
    }
    else
    {
        Serial.println("Failed to connect!");
        return;
    }

    
    peripheral.discoverAttributes();
    
    BLECharacteristic bleImuChar = peripheral.characteristic("F7580003-153E-D4F6-F26D-43D8D98EEB13");
    
    if (!bleImuChar)
    {
        peripheral.disconnect();
        Serial.println("Peripheral does not have IMU characteristic!");
        delay(5000);
        return;
    } 
      bleImuChar.subscribe();
      

    discovered = false;
    while (peripheral.connected())
    {
        if (bleImuChar.valueUpdated())
        {
            const unsigned char *cvalue = bleImuChar.value();
            const imuFrameType *value = (const imuFrameType *)cvalue;
            Serial.print("\r\nCharacteristic event, written: ");
            Serial.print(value->index);
            Serial.print("\t");
            Serial.print(value->slot[0]);
            Serial.print("\t");
            Serial.print(value->slot[1]);
            Serial.print("\t");
            Serial.println(value->slot[2]);

  // print the RSSI
    Serial.print("RSSI: ");
    Serial.println(peripheral.rssi());
        }
    }
    Serial.print("Disconnected");
    Serial.println(peripheral.address());
}

..................................................................
In that code I was just trying to measure the rssi of same peripheral the central is connected to and receives information regarding the Accelometer and gyro sensors.

but it doesn't work. it gives only same value. the peripheral has to be disconnected from the central so that the rssi to be rightly measured. Is there any explanation or solution for that?.

SidLeung, 2017-01-13:
The API, rssi(), is not functional yet with this release.  Will be looking into it.

DomiKoe, 2017-01-31:
Hi, are there any updates on this topic?

kitsunami, 2017-02-06:
This is not scheduled to be released in our next update.. it may be considered for future updates later this year.

janunezc, 2017-05-19:
Hi, any news on this one?

SidLeung, 2017-05-26:
Please note that RSSI information is not available once connection is established and, thus, it is beyond the scope of the BLE stack implementation.