arduino/ArduinoCore-arc32 #399: CDCSerial not available after PC reboot or resume from suspend

UDOOboard, 2017-01-13:
Leaving the Arduino 101 connected to the PC (as for an embedded installation), the Serial communication over USB doesn't work properly at reboot or resume from suspend state.
Both in Windows or Linux the COM or tty is shown on reboot, but it doesn't allow you to write data from PC to Arduino101 trought the Serial.

for example using the Serial Event Example:
```
/*
  Serial Event example

 When new serial data arrives, this sketch adds it to a String.
 When a newline is received, the loop prints the string and
 clears it.

 A good test for this is to try it with a GPS receiver
 that sends out NMEA 0183 sentences.

 Created 9 May 2011
 by Tom Igoe

 This example code is in the public domain.

 http://www.arduino.cc/en/Tutorial/SerialEvent

 */

String inputString = "";         // a string to hold incoming data
boolean stringComplete = false;  // whether the string is complete

void setup() {
  // initialize serial:
  Serial.begin(115200);
  while (!Serial) ;
  // reserve 200 bytes for the inputString:
  inputString.reserve(200);
}

void loop() {
  // print the string when a newline arrives:
  if (stringComplete) {
    Serial.println(inputString);
    // clear the string:
    inputString = "";
    stringComplete = false;
  }
}

/*
  SerialEvent occurs whenever a new data comes in the
 hardware serial RX.  This routine is run between each
 time loop() runs, so using delay inside loop can delay
 response.  Multiple bytes of data may be available.
 */
void serialEvent() {
  while (Serial.available()) {
    // get the new byte:
    char inChar = (char)Serial.read();
    // add it to the inputString:
    inputString += inChar;
    // if the incoming character is a newline, set a flag
    // so the main loop can do something about it:
    if ((inChar == '\n') || (inChar == 10) || (inChar == 13)) {
      stringComplete = true;
    }
  }
}
```
At reboot or resume from suspend state the Serial communication doesn't work properly and Arduino IDE freezes often while trying to write data on Serial through Serial Monitor.

Only a Master Reset resolve this issue.

fmntf, 2017-02-03:
@kitsunami is there a release scheduled to get this bug fixed?

kitsunami, 2017-02-06:
@bigdinotech Can you first try this on an Uno?

fmntf, 2017-02-07:
I tried it on Uno, it works fine.

IeiuniumLux, 2020-03-15:
Will this ever be released?