arduino/ArduinoCore-arc32 #408: Typecast issue when converting example code from .ino to .cpp

ajhool, 2017-01-19:
Hi, I am having trouble understanding the BLECharacteristic templates, and implementing the .ino example code in .cpp. Specifically, adding a typed BLECharacteristic (eg. BLEUnsignedLongCharacteristic) to a service. I have also reviewed the .cpp code that the Arduino IDE generates, and find that XCode IDE still won't compile it. Using master head of CurieBLE library Code and compiler error below (which follows patterns in the LED and LEDCallback examples provided):

`
BLEPeripheral blePeripheral;

BLEService service = BLEService("8888888-5555-1234-5678-D104768A1214");
BLEUnsignedCharCharacteristic failedChar = BLEUnsignedCharCharacteristic("8888888-5555-1235-5678-D104768A1214", BLERead | BLEWrite);
BLECharacteristic workingChar = BLECharacteristic(""8888888-5555-1236-5678-D104768A1214"", BLERead | BLEWrite , 1);

int main(int argc, const char * argv[]) {
    service.addServiceToPeripheral(blePeripheral);

    service.addCharacteristic(workingChar);

    //throws error in XCode: 
    //non-const !value reference to type 'BLECharacteristic' cannot bind to value of unrelated type 'BLEUnsignedLongCharacteristic
    service.addCharacteristic(failedChar);

}
`

As an aside, perhaps examples could be given in .cpp and .ino?

eriknyquist, 2017-01-23:
Hi @ajhool, corelibs-arduino101 is specifically for the Arduino IDE. I've never tried it, but I don't think it will work to just move the library files into Xcode.

At any rate, we need more information to help you;

1) Which compiler flags are being used in Xcode? can you paste them here please?
2) Please describe, in detail, the process you used to move the files into Xcode.

eriknyquist, 2017-02-08:
@ajhool ping, can you provide us with that information?

closed 2017-02-24 by kitsunami