arduino/ArduinoCore-arc32 #409: CurieBLE: Only get half of supposed advertising packets from ble_scan

Firni11, 2017-01-26:
I am currently working on BLE and we are very interested in the nRF Chip which is implemented on the Genuino 101. For our application we are looking for the best BLE Chip to track beacons (We just want to know the beacons address and the RSSI value).
During the last months we compared the onboard BLE Chip of the Raspberry Pi 3 with severel different BLE USB Dongles. For our last comparison we want some data from the nRF51822 Chip. To get Data from this we bought the Genuino 101.
To simply scan for beacons I uploaded the *ble_scan.ino* sketch from the 2.0.0 library to my Genuino 101.
```
#include <CurieBLE.h>

void setup() {
  Serial.begin(115200);

  // initialize the BLE hardware
  BLE.begin();

  Serial.println("BLE Central scan");

  // start scanning for peripheral
  BLE.scan();
}

void loop() {
  // check if a peripheral has been discovered
  BLEDevice peripheral = BLE.available();

  if (peripheral)
  {
       // discovered a peripheral
    
      // print address
      Serial.print(peripheral.address()+";");

      // print the local name, if present
      if (peripheral.hasLocalName())
      {
        Serial.print("Local Name: ");
        Serial.print(peripheral.localName()+";");
      }

      // print the advertised service UUID's, if present
      /*if (peripheral.hasAdvertisedServiceUuid())
      {
        Serial.print("Service UUID's: ");
        for (int i = 0; i < peripheral.advertisedServiceUuidCount(); i++)
        {
          Serial.print(peripheral.advertisedServiceUuid(i));
          Serial.print(" ");
        }
        Serial.println();
      }*/

      // print the RSSI
      Serial.print(peripheral.rssi());

      Serial.println();
    }
}
```
To summarize my collected data from the Genuino 101 (nRF51822), I can say that the there is much less noise in comparison with the data from the onboard Raspberry Pi Chip or USB Dongles.

**But** now I want to mention my current problem with the Genuino.
When I scan for beacons and my beacons advertising interval is set to 100ms I usually should receive 10 advertising packets per second with my Genuino. And for a 500ms ad-interval I should receive 2 packets per second. But at this time I only receive half of the amount of packets. So at a 100ms ad-interval I receive about 5 Packs/sec, and so on.
Does anyone of you know why this happens or is it possible to increase the ?scan-intervall? or something like that so I get every Packet which is transmitted by the beacon?

I would be really grateful If someone of you could help me.

Firni11, 2017-02-01:
Hi
Up to now I have researched on github for similar issues and I found out that my problem has something to do with the scan response (which has its own issue). I added the changes from that solution and now I get the right amount of packets. 
Sorry for wasting your time.

sandeepmistry, 2017-02-07:
@kitsunami @SidLeung any reason for closing this?

I think we should re-open it until scan responses are support by the library.

SidLeung, 2017-02-07:
As per the author, the issue seems to be resolved and, thus, closed.

SidLeung, 2017-02-07:
This issue could be related to the processing of the Scan Response Data message, [https://github.com/01org/corelibs-arduino101/issues/370](url)

Reopen this issue until #370 is addressed.

SidLeung, 2017-03-01:
The support for Scan Response Data req/rep will be in the next release.  PR #412 and PR #458 address this issue.

SidLeung, 2017-03-14:
The support for BLE Scan Respond Data request and reply is in the release 2.0.1 and, thus, closing out this PR.

closed 2017-03-14 by SidLeung