arduino/ArduinoCore-arc32 #413: Create a Hardware PWM library

LordDanielW, 2017-01-27:
CurieTimerOne is a great software solution to creating a PWM, but it limits the full potential of what could be achieved with the 4 quark hardware PWM's.  Creating a class for this would allow the use of 4 PWM's vice 1 and frequency up to 32MHz vice 100kHz.  I would be willing to create this library, but am currently stuck on how to change the firmware setting from GPIO_SS[10] to PWM[0].  Any help would be appreciated.

bigdinotech, 2017-01-27:
@LordDanielW take a look at the code for analogWrite() inside wiring_analog.c
You would see the code for muxing the gpio into pwm and also changing the clock speed.

eriknyquist, 2017-01-31:
@LordDanielW in case you don't already have it, the datasheet for the Quark SE C1000 (the SoC inside the Curie Module) may be helpful to you, if you are writing a new library.

http://www.intel.com/content/dam/www/public/us/en/documents/datasheets/quark-c1000-datasheet.pdf

FYI, we do try and use this issue tracker only for software defects. In future, if you could keep discussions like this in the Arduino forum, we would appreciate it! (we hang out there, too).

LordDanielW, 2017-02-01:
@bigdinotech Thank you so much, I have been looking everywhere for that.  

@eriknyquist  I have the datasheet, and am using the forums.  Neither of those had a solution to this problem.  I believe this issue tracker is meant for software defects and suggested features.  If this is not the case, please update the Support & Issues  part of readme on this repository.

I believe the best way to implement this is by adding a frequency variable to analogWrite.  I would be willing to implement this, or create a new function.  If that is not the direction that you guys want this to go though, my project is running now and I would be okay with closing this issue.

void analogWrite(uint8_t pin, uint32_t val, uint32_t freq)

PaulStoffregen, 2017-02-01:
Years ago, after much public discussion, @cmaglie decided the Arduino API would someday add analogWriteFrequency(pin, frequency).  So far, it seems I'm the only one who's ever implemented this for any board.  Perhaps it's too far in the past, now long forgotten?  But even if something different is now decided, I would hope this can be done with Cristian and the rest of the Arduino Team's consent to make it an official part of the API and implement a compatible API for all boards.  It's more work, but long-term the whole Arduino world benefits if we can all do things in ways that have compatible APIs across different hardware.

LordDanielW, 2017-02-01:
@PaulStoffregen, it looks like bigdinotech implemented that exact function in the latest beta of arduino 101.  Thanks again for all the help.

closed 2017-02-01 by LordDanielW