arduino/ArduinoCore-arc32 #415: Curie BLE for Edison?

hahnicity, 2017-01-27:
Hi, my main question is; is there a way to have Curie BLE be functional on the Intel Edison board? I'm an Arduino newb but I'd be happy to help create additional functionality for CurieBLE if it's possible to do so.

kitsunami, 2017-01-30:
Currently Edison doesn't have this support, but we'll pass this along to the Edison team.

hahnicity, 2017-01-30:
Would the obstacle to implementation be hardware related? Or is it a software issue?

SidLeung, 2017-01-31:
The BLE stack that runs on the Arduino 101 board is developed for the specific BLE h/w in the Curie SoC.  The stack can be divided into 3 layers:  API, link layer, and the hardware specific BLE stack.  For starter, the Edison BLE h/w is different from the Arduino and that would make the hardware layer not applicable.

Without much understanding of the Edison system, I would think that the Bluetooth 4.0 stack is integrated into the Linux OS and generic Bluetooth apps should be able to run on that platform.  If Edison Bluetooth stack supports dual mode, Classic and BLE, BLE applications should be able to run on that platform to interface with an Arduino board.

hahnicity, 2017-01-31:
OK thank you for the giving me the basics on that!

closed 2017-02-06 by kitsunami