arduino/ArduinoCore-arc32 #420: CurieBLE: add broadcast property to characteristic

don, 2017-02-01:
I was porting some [BLEPeripheral](https://github.com/sandeepmistry/arduino-BLEPeripheral) code to the Arduino 101 and noticed that BLEBroadcast is missing from CurieBLE.

[image]

I'd like to be able to add the [BLEBroadcast property](https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/API.md#constructor-2) to a Characteristic. [setValue](https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/API.md#set-value) should update the advertising data when the characteristic's value changes.

SidLeung, 2017-02-01:
Hi, Don:

I have not use the broadcast() API before in the previous library.  However, I have used the BLENotify option in a Characteristic in one of my Peripheral sketches.  To notify the connected Central for data ready, my sketch just need to use the setValue() API to set the local buffer with the data.  The BLE library takes care of the notification process.  There is no specific API to call to prompt the library to perform a notification.  I would think the broadcast option to be similar.

don, 2017-02-02:
@SidLeung BLENotify requires a connection between the client and the server.

BLEBroadcast allows me to send data without a connection. I created a [demo sketch](https://gist.github.com/don/536d9f1d605d9bfffcd99b5716a72cb4) for you. I ran it on a RFduino, but it'll work on RedBear Nano or other boards supported by [BLEPeripheral](https://github.com/sandeepmistry/arduino-BLEPeripheral#compatible-hardware). The peripheral increments a counter and sets the characteristic value. Since the characteristic is marked as broadcast, the advertising data is updated whenever the value changes. This allows a my phone to get the counter value when scanning. I don't need to connect to get the data.

    // Broadcast Characteristic Value
    // Simple counter that broadcasts a value
    #include <BLEPeripheral.h>

    uint8_t value = 0;

    BLEPeripheral peripheral;
    BLEService service = BLEService("EEE0");
    BLEShortCharacteristic characteristic = BLEShortCharacteristic("EEE1", BLERead | BLENotify | BLEBroadcast);

    void setup() {
      Serial.begin(9600);

      peripheral.setLocalName("BLEBroadcast");
      peripheral.setAdvertisedServiceUuid(service.uuid());
  
      peripheral.addAttribute(service);
      peripheral.addAttribute(characteristic);
  
      characteristic.setValue(value);

      peripheral.begin();
      characteristic.broadcast();

      Serial.println(F("BLE Broadcast Count"));
    }

    void loop() {
        peripheral.poll();
        characteristic.setValue(value);    
        delay(1000);
        value++;
    }
https://gist.github.com/don/536d9f1d605d9bfffcd99b5716a72cb4
[image]

SidLeung, 2017-02-02:
Have created a Jira ticket internally to look into BLE broadcasting in the 101 stack.  Will post any result here.

don, 2017-02-02:
@SidLeung thanks!

sgbihu, 2017-02-13:
@don 
I test your code on CurieBLE lib. Just change the include head file, the program can works.

Please change this line
#include <BLEPeripheral.h>
as
#include <CurieBLE.h>

The sketch can works.

SidLeung, 2017-02-13:
@don , is it possible to use your sketch as an example as part of the coreLibs release?  If so, please add an licensing header and we will pull it into the next release.  Thanks.

calvinatintel, 2017-02-14:
Hi @don. Thank you for your contribution. We love your sample code and have created a PR https://github.com/01org/corelibs-arduino101/pull/437 with it. We need your written approval that we can include your code in our package, in the form of a software license. Arduino 101 CoreLibs is distributed under LGPLv2.1+ license and you can choose any software license that is compatible with LGPLv2.1+. According to [Wikiepedia](https://en.wikipedia.org/wiki/Free_software_license), it can be

* Public domain
* MIT / X11
* BSD (revised)
* Apache 2.0
* or LGPL 2.1+ itself

Please let me know once you've decided on a license. I can help you apply it onto the code. Thank you.

closed 2017-02-24 by kitsunami

ElliotMebane, 2017-05-12:
EddystoneURL sample with Curie 2.0.2:
https://github.com/ElliotMebane/Arduino/tree/master/EddystoneURL_Arduino101