arduino/ArduinoCore-arc32 #422: Arduino 101 EEPROM.h included more than once produces compile errors

drmpf, 2017-02-01:
If you 
#include <EEPROM.h>
in more then one file, e.g. in the main sketch and in a library file or another .ino file
Then the compiler exits with errors like

Arduino15\packages\Intel\hardware\arc32\1.0.7\libraries\EEPROM\src/EEPROM.h:38: multiple definition of `CurieClear()'

due to EEPROM.h have method code in the header file.  

Moving the methods to a EEPROM.cpp file and declaring them at the top of the EEPROM.h file seems to fix the problem.

eriknyquist, 2017-02-01:
@drmpf, you are correct, the way to fix this is to put only the function declarations in the header file, and make a new .cpp file for the actual definitions.

The include guards will then work as expected. Thanks for pointing this out, we'll fix it soon :)

intel-larry, 2017-02-01:
Agree - just ran into this yesterday trying to compile the Arducam code. Was just about to write an issue on it. drmpf is correct.

eriknyquist, 2017-02-02:
Fixed in https://github.com/01org/corelibs-arduino101/pull/425

closed 2017-02-09 by eriknyquist