arduino/ArduinoCore-arc32 #428: CurieBLE: Won't advertise when initiatizing multiple services

acaballero, 2017-02-06:
Hello,
I'm trying to create two services, a standard battery service and a custom one. When I create only the custom service, everyting goes right, but If I create two, the device won't even announce itself. 

I'm doing like this:

```
   BLEPeripheral blePeripheral;  // BLE Peripheral Device (the board you're programming)
   BLEService controlService("19B10000-E8F2-537E-4F6C-D104768A1214"); // BLE Control Service
   BLEService batteryService("180F"); // BLE Control Service

   // BLE LED Switch Characteristic - custom 128-bit UUID, read and writable by central
   BLECharacteristic switchCharacteristic("19B10001-E8F2-537E-4F6C-D104768A1214", BLERead | BLEWrite, 20);
   // BLE Tx Characteristic for the central to read responses
   BLECharacteristic txCharacteristic("19B10002-E8F2-537E-4F6C-D104768A1214", BLENotify | BLERead, 20);
   // BLE Rx Characteristic for the central to write commands
   BLECharacteristic rxCharacteristic("19B10003-E8F2-537E-4F6C-D104768A1214", BLEWrite, 20);
   // BLE characteristic for the device to publish it's battery level
   BLECharacteristic batteryCharacteristic("0A19", BLENotify | BLERead, 1);

   BLEDescriptor ledDescriptor = BLEDescriptor("2901", "Led status");
   BLEDescriptor txDescriptor = BLEDescriptor("2901", "TX");
   BLEDescriptor rxDescriptor = BLEDescriptor("2901", "RX");
   BLEDescriptor batteryDescriptor = BLEDescriptor("2901", "Battery");

    blePeripheral.setDeviceName(DEVICE_NAME);
    blePeripheral.setLocalName(DEVICE_NAME);
    blePeripheral.setAppearance(0x0180);

    blePeripheral.setAdvertisedServiceUuid(controlService.uuid());
    blePeripheral.setAdvertisedServiceUuid(batteryService.uuid());

    // add service and characteristic:
    blePeripheral.addAttribute(controlService);

    blePeripheral.addAttribute(switchCharacteristic);
    blePeripheral.addAttribute(ledDescriptor);
    blePeripheral.addAttribute(txCharacteristic);
    blePeripheral.addAttribute(txDescriptor);
    blePeripheral.addAttribute(rxCharacteristic);
    blePeripheral.addAttribute(rxDescriptor);

    blePeripheral.addAttribute(batteryService);

    blePeripheral.addAttribute(batteryCharacteristic);
    blePeripheral.addAttribute(batteryDescriptor);
```

Looking at the BLEPeripheral code, it seems that it only takes in account one service. Is it like it seems?

Thanks in advance!

noelpaz, 2017-02-07:
There is a limitation in the total number of attributes characteristics, descriptors as well as services, although I think that you should not fail with what you have more. I am not seeing the whole code are there more services and characteristics? Any the begin function should return a true or false and if it return falls it should return false. Can you include the whole code so I could replicate your problem?

SidLeung, 2017-02-08:
There is a limitation on the number of Characteristic can be added.  This is due to memory constraint.  However, we will review this restriction before the next release, especially in the case of Peripheral.

acaballero, 2017-02-08:
It's working now. I was confused about 'setAdvertisedServiceUuid', and it turns out it's supposed to be called only once, as it assings the *main* service Uuid associated with the peripheral. You can later add more services following the correct order, which is:

service
   characteristic
      [descriptor, ...]
   [characteristic,..]
[service, ....]

Doing like that, the device announces all the services, even if there's only one *main* service Uuid, the one you have assigned by calling 'setAdvertisedServiceUuid'.

So, regarding the code, it's working by removing the second call to 'setAdvertisedServiceUuid':

`   blePeripheral.setDeviceName(DEVICE_NAME);
    blePeripheral.setLocalName(DEVICE_NAME);
    blePeripheral.setAppearance(0x0180);

    blePeripheral.setAdvertisedServiceUuid(controlService.uuid());
    
// Only one Uuid should be assigned by calling setAdvertisedServiceUuid
// blePeripheral.setAdvertisedServiceUuid(batteryService.uuid());

    // add service and characteristic:
    blePeripheral.addAttribute(controlService);

    blePeripheral.addAttribute(switchCharacteristic);
    blePeripheral.addAttribute(ledDescriptor);
    blePeripheral.addAttribute(txCharacteristic);
    blePeripheral.addAttribute(txDescriptor);
    blePeripheral.addAttribute(rxCharacteristic);
    blePeripheral.addAttribute(rxDescriptor);

    blePeripheral.addAttribute(batteryService);

    blePeripheral.addAttribute(batteryCharacteristic);
    blePeripheral.addAttribute(batteryDescriptor);
`

Thanks!

closed 2017-02-08 by acaballero

noelpaz, 2017-02-09:
Yes correct in 1.0.7 , you have to add the characteristics below the service and characteristic descriptors below the characteristics. It's a herirarchy