arduino/ArduinoCore-arc32 #432: CurieIMU with interrupt changes local variables in other function.

jimaobian, 2017-02-08:
It is really an odd issue.
As shown in the loop function. All the value in test array should equal to sumX. However sometimes it will not be true. This problem doesn't always happen. Its ratio is about 40%.

I'm not sure whether it is a good idea to get acceleration value in interrupt context. But in theory, interrupt may not change the local variables in other function.

[image]

Here is my test code:

```
#include "CurieIMU.h"

void interruptIMU(){

  int16_t accX = 0;
  int16_t accY = 0;
  int16_t accZ = 0;
  
  CurieIMU.getAcceleration(&accX, &accY, &accZ);
}

void setup()
{
  Serial.begin(115200);
  while(!Serial);

  CurieIMU.begin();
  CurieIMU.interrupts(CURIE_IMU_DATA_READY);
  CurieIMU.attachInterrupt(interruptIMU);
}

void loop ()
{
  int16_t test[10];
  int16_t sumX = 0;
  for (int i=0; i<10; i++) {
    test[i] = sumX;
  }
  
  if (test[9]!=sumX) {
    Serial.println("This should never happen");
  }
}
```

Environment:
OS: Mac OSX 10.12.3
Arduino: 1.8.1
Curie Package: 1.0.7

eriknyquist, 2017-02-08:
Thanks for reporting, @jimaobian. Can you give me an idea of how long you had the sketch running, in order to see "This should never happen"?

eriknyquist, 2017-02-08:
I've been running it for 2 hours, nothing yet. I'll leave it overnight.

SidLeung, 2017-02-08:
@jimaobian 

You are correct.  Interrupt context could not impact the background context.  Before going into this subject, a minor suggestion first:  It is a good practice to configure the interrupt event and its call back handler prior to kicking off the operation.  The suggested changes are,

  CurieIMU.interrupts(CURIE_IMU_DATA_READY);
  CurieIMU.attachInterrupt(interruptIMU);
  CurieIMU.begin();

Back to the interrupt issue.  During the development of the BLE library, we recently fixed a nasty bug in the ISR.  The loop register (a special purpose h/w register in the ARC core) was not saved onto the interrupt stack during context switching.  As a consequence, background code that used this looping register (determined by the compiler) would be affected by the foreground (interrupt context) code that also used this looping register.

Not sure if you ran into this bug with your sketch.  However, you could do an experiment with your sketch.  You can replace the for-loop that initializes test[] with just plain slot by slot initialization as follow,

  test[0] =
  test[1] =
    ...
  test[9] = 0;

The purpose of this change is to eliminate the use of looping register by the compiler.  Would like to hear the result of this experiment from you soon.

eriknyquist, 2017-02-09:
@jimaobian Everything still running correctly for me after 6 hours.... have not yet seen the message

jimaobian, 2017-02-09:
@eriknyquist Try to upload this sketch many times or just reset it. It happens to me once every two or three tries. For this example, this issue seems only happening at the beginning. And I have another big example, which will always trigger it.
```
/*
 * This example demonstrates using the pattern matching engine (CuriePME)
 * to classify streams of accelerometer data from CurieIMU.
 *
 * First, the sketch will prompt you to draw some letters in the air (just
 * imagine you are writing on an invisible whiteboard, using your board as the
 * pen), and the IMU data from these motions is used as training data for the
 * PME. Once training is finished, you can keep drawing letters and the PME
 * will try to guess which letter you are drawing.
 *
 * This example requires a button to be connected to digital pin 4
 * https://www.arduino.cc/en/Tutorial/Button
 *
 * NOTE: For best results, draw big letters, at least 1-2 feet tall.
 *
 * Copyright (c) 2016 Intel Corporation.  All rights reserved.
 * See license notice at end of file.
 */

#include "CurieIMU.h"
#include "CuriePME.h"

/*  This controls how many times a letter must be drawn during training.
 *  Any higher than 4, and you may not have enough neurons for all 26 letters
 *  of the alphabet. Lower than 4 means less work for you to train a letter,
 *  but the PME may have a harder time classifying that letter. */
const unsigned int trainingReps = 4;

/* Increase this to 'A-Z' if you like-- it just takes a lot longer to train */
const unsigned char trainingStart = 'A';
const unsigned char trainingEnd = 'F';

/* The input pin used to signal when a letter is being drawn- you'll
 * need to make sure a button is attached to this pin */
const unsigned int buttonPin = 4;

/* Sample rate for accelerometer */
const unsigned int sampleRateHZ = 200;

/* No. of bytes that one neuron can hold */
const unsigned int vectorNumBytes = 128;

/* Number of processed samples (1 sample == accel x, y, z)
 * that can fit inside a neuron */
const unsigned int samplesPerVector = (vectorNumBytes / 3);

/* This value is used to convert ASCII characters A-Z
 * into decimal values 1-26, and back again. */
const unsigned int upperStart = 0x40;

const unsigned int sensorBufSize = 2048;
const int IMULow = -32768;
const int IMUHigh = 32767;

#define BUFFER_LENGTH 85

uint8_t rawIndex = 0;
uint8_t rawX[BUFFER_LENGTH] = {0};
uint8_t rawY[BUFFER_LENGTH] = {0};
uint8_t rawZ[BUFFER_LENGTH] = {0};

void trainLetters();
void readVectorFromIMU(byte vector[]);

uint8_t dataBuffer[BUFFER_LENGTH*3];

#define IDLE_AVERAGE_NUM 10
bool isIdle(unsigned long duration){
[... 342 lines cut ...]
    Serial.print(String(i) + "' in the air. Release the button as soon ");
    Serial.println("as you are done.");
    
    trainLetter(i, trainingReps);
    Serial.println("OK, finished with this letter.");
    delay(2000);
  }
}

/*
 * This library is free software; you can redistribute it and/or
 * modify it under the terms of the GNU Lesser General Public
 * License as published by the Free Software Foundation; either
 * version 2.1 of the License, or (at your option) any later version.
 *
 * This library is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 * Lesser General Public License for more details.
 *
 * You should have received a copy of the GNU Lesser General Public
 * License along with this library; if not, write to the Free Software
 * Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301 USA
 */
```

jimaobian, 2017-02-09:
@SidLeung Awesome! Your code solve my problem :). Many thanks.
```
CurieIMU.interrupts(CURIE_IMU_DATA_READY);
CurieIMU.attachInterrupt(interruptIMU);
CurieIMU.begin();
```
Continue to wait for the fix of the interrupt issue.

jimaobian, 2017-02-09:
Updated:
Here is the reason why the short example only show the "This should never happen" at the beginning:
The print "Hello" code in loop is not running every 0.5s. It blocks after 2s or 3s. Here is the code.
```
#include "CurieIMU.h"

void interruptIMU(){

  int16_t accX = 0;
  int16_t accY = 0;
  int16_t accZ = 0;
  
  CurieIMU.getAcceleration(&accX, &accY, &accZ);
}

void setup()
{
  Serial.begin(115200);
  while(!Serial);

  CurieIMU.begin();

  CurieIMU.interrupts(CURIE_IMU_DATA_READY);
  CurieIMU.attachInterrupt(interruptIMU);

}

void loop ()
{
  int16_t test[10];
  int16_t sumX = 0;

  for (int i=0; i<10; i++) {
    test[i] = sumX;
  }

  if (test[9]!=sumX) {
    Serial.println("This should never happen");
  }

  static unsigned long timer = millis();
  if(millis() - timer>500){
    timer = millis();
    Serial.println("Hello");
  }
}
```

Change the order like the following will not help, it also prevents the loop function from running:
```
CurieIMU.interrupts(CURIE_IMU_DATA_READY);
CurieIMU.attachInterrupt(interruptIMU);
CurieIMU.begin();
```

jimaobian, 2017-02-09:
Update to the latest version deal with the problem.
Thank you.

closed 2017-02-09 by jimaobian

SidLeung, 2017-02-09:
@jimaobian ,

My apology for giving out poor advise and hopefully did not confuse more people.  The  CurieIMU::begin needs to be called prior to attaching interrupts and event handler.  My fault for advising the reverse order.  If you take a quick look at begin, you will find that it completes the construction of the CurieIMU object (with resources allocation) rather then just kicking off the IMU unit.  Thus, cannot install event handler prior to construction completion.