arduino/ArduinoCore-arc32 #441: How does auto-calibration work in this specific case? Why the offset can be assumed to be linear?

uomodellamansarda, 2017-02-15:
How does auto-calibration work in this specific case? This is not very clear to me.
  Why the offset can be assumed to be linear?
  Is there any reference stating that it is safe to assume a linear offset for 1 axis calibration?

The CurieIMUgetAccelerometerOffset reference says that the sensitivity is 3,9 mg.

I have asked on StackOverflow [here](http://stackoverflow.com/questions/42118941/how-does-genuino-arduino-101-intel-curie-autocalibration-ax-ay-az-process-work) before posting, and waited for days. 

Given a sensitivity of +-2g, I have to use this formula

3,9*19,6/32767

in order to convert the value in m/s^2.

Here is the library code that computes the correct offset:

     {
         int bmiOffset = offset / 3.9;

    if (bmiOffset < -128) {
        bmiOffset = -128;
    } else if (bmiOffset > 127) {
        bmiOffset = 127;
    }

    if (axis == X_AXIS) {
        setXAccelOffset(bmiOffset);
    } else if (axis == Y_AXIS) {
        setYAccelOffset(bmiOffset);
    } else if (axis == Z_AXIS) {
        setZAccelOffset(bmiOffset);
    }

    setAccelOffsetEnabled(true);
    }

eriknyquist, 2017-02-15:
@uomodellamansarda I'm not sure offhand, as I didn't write this code myself, but I imagine the datasheet for the BMI160 will have the answer you're looking for

http://www.mouser.com/ds/2/783/BST-BMI160-DS000-07-786474.pdf

eriknyquist, 2017-02-15:
Oh, and for any future questions you have, the Arduino 101 forum is a good place to go, a lot of people hang out there so you should get pretty quick responses :)

https://forum.arduino.cc/index.php?board=103.0

uomodellamansarda, 2017-02-21:
I posted on the forum too, [in the Sensors section ](https://forum.arduino.cc/index.php?topic=456595.0)
I don't know if is the correct section or is better in Arduino 101, i will not create duplicate.
But no one still answered.
Thanks :)

eriknyquist, 2017-02-22:
Well, the short answer is, I do not know the answer to your question. You could try having your post moved to the rduino101 section, I suspect you may get more responses then.

This issue tracker is for reporting software defects, so I'm going to close this issue.

closed 2017-02-22 by eriknyquist