arduino/ArduinoCore-arc32 #444: CurieBLE: BLEPeripheral::connected() always returns false

sandeepmistry, 2017-02-17:
Observed while debugging: https://github.com/01org/corelibs-arduino101/issues/377#issuecomment-280759093

```arduino
#include <CurieBLE.h>

BLEPeripheral blePeripheral;

void setup() {
  Serial.begin(9600);
  while(!Serial);

  blePeripheral.setLocalName("test");

  blePeripheral.begin();
}

void loop() {
  if (blePeripheral.connected()) {
    Serial.println("connected");

    while (blePeripheral.connected());

    Serial.println("disconnected");
  }

}
```

if i connect from a central, "connected" is never printed.

sgbihu, 2017-02-20:
Fixed.

SidLeung, 2017-02-22:
Jira ticket 857 is assigned to track this issue.

russmcinnis, 2017-02-22:
We ran a round of BAT tests along with some manual things that require close timing and everything passes. The test sketch here is showing connected when running against nRF on Android phone. So it looks like the return value has been fixed.

sandeepmistry, 2017-03-02:
I tested with Github master d818a3a5fa1ede20004487ffc5c95e6008b59a8d, this is fixed. Closing.

closed 2017-03-02 by sandeepmistry