arduino/ArduinoCore-arc32 #445: CurieBLE: Deprecated BLEConnected and BLEDisconnected events not triggered

sandeepmistry, 2017-02-17:
While testing the v1.0.7 LED callback example (https://github.com/01org/corelibs-arduino101/blob/1.0.7/libraries/CurieBLE/examples/CallbackLED/CallbackLED.ino), I noticed the BLEConnected and BLEDisconnected events are not being trigged.

sgbihu, 2017-02-20:
Fixed

SidLeung, 2017-02-22:
This issue is tracked by Jira ticket 858.

russmcinnis, 2017-02-22:
Confirmed that this is working with PR447 using the sketch linked here. Also ran basic BAT tests along with some manual testing.

russmcinnis, 2017-02-27:
this works with PR 455 system tests and tested for seeing connects and disconnects. Looks good.

sandeepmistry, 2017-03-15:
Tested again with the 2.0.1 RC1 release, this now works as expected. Closing.

```
Connected event, central: AC:BC:32:7C:FA:26
Characteristic event, written: LED on
Characteristic event, written: LED off
Disconnected event, central: AC:BC:32:7C:FA:26
```

closed 2017-03-15 by sandeepmistry