arduino/ArduinoCore-arc32 #446: CurieBLE: Adding more than 6 characteristics to a service causes device to crash when attempting to advertise

kgillespieanaren, 2017-02-18:
It appears there is an issue when you add more than 6 characteristics to a single service in the CurieBLE driver with version 2.0.0 RC4.

The device never begins to advertise and appears to hang when calling BLE.advertise();

Removing the 7th characteristic from service will allow the device to continue to work.

Using Arduino101 board and version 2.0.0 RC4 of the Intel Curie Arduino SDK libs.

kgillespieanaren, 2017-02-18:
`
#include <CurieBLE.h>
#include "CurieTimerOne.h"

#ifndef INTERVAL_TIMER
#define INTERVAL_TIMER 1000
#endif

static bool intervalFireOff = false;

BLEService CloudPrimaryService("bfe433cf-6b5e-4368-abab-b0a59666a402");
BLECharacteristic CloudCredentials("bfe433cf-6b5e-4368-abab-b0a59666a403", BLERead | BLEWrite, 64);
BLECharacteristic CloudNotify("bfe433cf-6b5e-4368-abab-b0a59666a404", BLERead | BLEWrite | BLENotify, 64);
BLEService PrimaryService("7c866f08-b22e-4294-a113-3839354957f6");
BLECharacteristic NotifyEvent("7c866f09-b22e-4294-a113-3839354957f6", BLERead | BLEWrite | BLENotify, 128);
BLECharacteristic Function1Characteristic("7c866f0a-b22e-4294-a113-3839354957f6", BLERead | BLEWrite, 0x1);
BLECharacteristic Function2Characteristic("7c866f0b-b22e-4294-a113-3839354957f6", BLERead | BLEWrite, 0x1);
BLECharacteristic Function3Characteristic("7c866f0c-b22e-4294-a113-3839354957f6", BLERead | BLEWrite, 0x1);
BLECharacteristic Function4Characteristic("7c866f0d-b22e-4294-a113-3839354957f6", BLERead | BLEWrite, 0x1);
BLECharacteristic Function5Characteristic("7c866f0e-b22e-4294-a113-3839354957f6", BLERead | BLEWrite, 0x1);
BLECharacteristic Function6Characteristic("7c866f0f-b22e-4294-a113-3839354957f6", BLERead | BLEWrite, 0x1);

void characteristicSubscribed(BLECentral& central, BLECharacteristic& characteristic) {
  // characteristic subscribed event handler
  //Serial.println(F("Characteristic event, subscribed"));
}

void characteristicUnsubscribed(BLECentral& central, BLECharacteristic& characteristic) {
  // characteristic unsubscribed event handler
  //Serial.println(F("Characteristic event, unsubscribed"));
}

void intervalTimerIsr() {
	intervalFireOff = true;
	
}

void setup() {
	BLE.begin();
	BLE.setLocalName("Test101");
	BLE.setAdvertisedServiceUuid(PrimaryService.uuid());
	BLE.setDeviceName("Atmosphere Test");
	// add service and characteristics

	CloudPrimaryService.addCharacteristic(CloudCredentials);
	CloudPrimaryService.addCharacteristic(CloudNotify);

	PrimaryService.addCharacteristic(NotifyEvent);

	PrimaryService.addCharacteristic(Function1Characteristic);
	PrimaryService.addCharacteristic(Function2Characteristic);
	PrimaryService.addCharacteristic(Function3Characteristic);
	PrimaryService.addCharacteristic(Function4Characteristic);
	PrimaryService.addCharacteristic(Function5Characteristic);
	
	//If you uncomment this line it'll no longer advertise
	//PrimaryService.addCharacteristic(Function6Characteristic);

	BLE.addService(PrimaryService);
	BLE.addService(CloudPrimaryService);

	CurieTimerOne.start(1000 * INTERVAL_TIMER, &intervalTimerIsr);
	BLE.advertise();
}

void doIntervalCheck()
{
	if(intervalFireOff)
	{
		intervalFireOff = false;
	}
}

void loop() {
	doIntervalCheck();
	
	// poll peripheral
	BLEDevice central = BLE.central();

	if(central) 
	{
		if(central.connected())
		{
			BLE.stopAdvertise();
			while(central.connected())
			{
				doIntervalCheck();

			}
			BLE.advertise();
		}
	}
}

`

kgillespieanaren, 2017-02-18:
[charTest.ino.zip](https://github.com/01org/corelibs-arduino101/files/785124/charTest.ino.zip)

In this example I have commented out the line

PrimaryService.addCharacteristic(Function6Characteristic);

If you uncomment it then this project will no longer run, and the Arduino101 won't advertise.

SidLeung, 2017-02-19:
Please note that the current memory arrangement can only provided 8 KB of HEAP for the entire system.  In addition, the malloc() does not maximize the usage of the HEAP.  This issue is under investigation and proposed solutions are under testing.  At the mean time, the number of Characteristic is limited.

kgillespieanaren, 2017-02-20:
Is there anyway to configure the heap to have more than 8KB allocated for a project?

SidLeung, 2017-02-21:
The allocation of ARC memory is controlled by the firmware running on the x86 core.  To re-arrange the allocation needs an upgrade to the firmware.

sandeepmistry, 2017-02-21:
> The device never begins to advertise and appears to hang when calling BLE.advertise();

@SidLeung is there are way to prevent the hanging?

kitsunami, 2017-02-21:
There is no solution to prevent the hanging at this time, but can be considered for future investigation/improvements.

agdl, 2017-03-29:
I have a sketch arranged like this:
```
/*BLE characterists and serices*/
BLEPeripheral blePeripheral;

BLEService service1("1E600000-xxxx-xxxx-xxxx-xxxxxxxxxxxx");
BLEUnsignedIntCharacteristic char1("1E600001-xxxx-xxxx-xxxx-xxxxxxxxxxxx", BLERead | BLEWrite | BLENotify);
BLEUnsignedCharCharacteristic char2("1E600002-xxxx-xxxx-xxxx-xxxxxxxxxxxx", BLERead | BLEWrite);
BLECharacteristic char3("1E600003-xxxx-xxxx-xxxx-xxxxxxxxxxxx", BLERead | BLEWrite | BLENotify, 20);
BLECharacteristic char4("1E600004-xxxx-xxxx-xxxx-xxxxxxxxxxxx", BLERead | BLEWrite | BLENotify, 20);

BLEService service2("2E600000-xxxx-xxxx-xxxx-xxxxxxxxxxxx");
BLEUnsignedIntCharacteristic char1("2E600001-xxxx-xxxx-xxxx-xxxxxxxxxxxx", BLERead | BLEWrite | BLENotify);
BLEUnsignedIntCharacteristic char2("2E600002-xxxx-xxxx-xxxx-xxxxxxxxxxxx", BLERead | BLEWrite | BLENotify);
BLEUnsignedIntCharacteristic char3("2E600003-xxxx-xxxx-xxxx-xxxxxxxxxxxx", BLERead | BLEWrite | BLENotify);
BLEUnsignedIntCharacteristic char4("2E600004-xxxx-xxxx-xxxx-xxxxxxxxxxxx", BLERead | BLEWrite | BLENotify);

BLEService service3("3E600000-xxxx-xxxx-xxxx-xxxxxxxxxxxx");
BLEUnsignedCharCharacteristic char1("3E600001-xxxx-xxxx-xxxx-xxxxxxxxxxxx", BLERead | BLEWrite);
BLEUnsignedCharCharacteristic char2("3E600002-xxxx-xxxx-xxxx-xxxxxxxxxxxx", BLERead | BLEWrite);
BLEUnsignedIntCharacteristic char3("3E600003-xxxx-xxxx-xxxx-xxxxxxxxxxxx", BLERead | BLEWrite);
/*End of BLE characterists and variables*/
```

That was working on 1.0.7

Now I can see only the services 0x1800 and 0x1801 and not mine.

Tested on 2.0.1

solidcloudio, 2017-04-15:
Seeing this same problem..    Sample script that fails:

`
#include <CurieBLE.h>

#define DEVICE_NAME "TestDevice"

#define SERVICE_UUID          ("f513d9a0-6cd5-4ed6-b98c-b67ce98f0cz4")

#define CHAR_UUID_1                 ("f513d9a1-6cd5-4ed6-b98c-b67ce98f0cz4")
#define CHAR_UUID_2                 ("f513d9a2-6cd5-4ed6-b98c-b67ce98f0cz4")
#define CHAR_UUID_3                 ("f513d9a3-6cd5-4ed6-b98c-b67ce98f0cz4")
#define CHAR_UUID_4                 ("f513d9a4-6cd5-4ed6-b98c-b67ce98f0cz4")
#define CHAR_UUID_5                 ("f513d9a5-6cd5-4ed6-b98c-b67ce98f0cz4")
#define CHAR_UUID_6                 ("f513d9a6-6cd5-4ed6-b98c-b67ce98f0cz4")
#define CHAR_UUID_7                 ("f513d9a7-6cd5-4ed6-b98c-b67ce98f0cz4")
#define CHAR_UUID_8                 ("f513d9a8-6cd5-4ed6-b98c-b67ce98f0cz4")
#define CHAR_UUID_9                 ("f513d9a9-6cd5-4ed6-b98c-b67ce98f0cz4")

bool connectedBle = false;       // Are we connected to a BLE device?

/* BLE Service */
BLEService pbService(SERVICE_UUID);

BLEUnsignedIntCharacteristic LvlChar1(CHAR_UUID_1, BLERead | BLENotify | BLEWrite);     
BLEUnsignedIntCharacteristic LvlChar2(CHAR_UUID_2, BLERead | BLENotify | BLEWrite);     
BLEUnsignedIntCharacteristic LvlChar3(CHAR_UUID_3, BLERead | BLENotify | BLEWrite);     
BLEUnsignedIntCharacteristic LvlChar4(CHAR_UUID_4, BLERead | BLENotify | BLEWrite);     
BLEUnsignedIntCharacteristic LvlChar5(CHAR_UUID_5, BLERead | BLENotify | BLEWrite);     
BLEUnsignedIntCharacteristic LvlChar6(CHAR_UUID_6, BLERead | BLENotify | BLEWrite);     
BLEUnsignedIntCharacteristic LvlChar7(CHAR_UUID_7, BLERead | BLENotify | BLEWrite);     

BLECharacteristic imuAccCharacteristic(CHAR_UUID_8, BLERead | BLENotify, 12);
BLECharacteristic imuGyroCharacteristic(CHAR_UUID_9, BLERead | BLENotify, 12);

void bleDeviceConnectHandler(BLEDevice central) {
    // central connected event handler
  Serial.print(F("Connected event, central: "));
  Serial.println(central.address());
  connectedBle = true;
}

void bleDeviceDisconnectHandler(BLEDevice central) {
    // central disconnected event handler
  Serial.print(F("Disconnected event, central: "));
  Serial.println(central.address());
  connectedBle = false;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  while (!Serial);    // wait for the serial port to open IF DEBUGGING!!
  
  Serial.println("Initializing BLE device...");

  /* Initialize BLE */
  BLE.begin();
  BLE.setLocalName("TestService");
  BLE.setAdvertisedService(pbService);

  // Add the characteristics
  pbService.addCharacteristic(LvlChar1);  
  pbService.addCharacteristic(LvlChar2);
  pbService.addCharacteristic(LvlChar3);
  pbService.addCharacteristic(LvlChar4);
  pbService.addCharacteristic(LvlChar5);
  pbService.addCharacteristic(LvlChar6);
  pbService.addCharacteristic(LvlChar7);
  
  // Add the characteristics
  int ret1 = pbService.addCharacteristic(imuAccCharacteristic);
  int ret2 = pbService.addCharacteristic(imuGyroCharacteristic);

  Serial.println(ret1);
  Serial.println(ret2);

  BLE.setEventHandler(BLEConnected, bleDeviceConnectHandler);
  BLE.setEventHandler(BLEDisconnected, bleDeviceDisconnectHandler);

  Serial.println("Events wired...");

  BLE.addService(pbService);              // Add the Imu service
  Serial.println("Service Added...");

  BLE.setTxPower(4);
  Serial.println("Power Set...");

  int bleStatus = BLE.advertise();
  Serial.println(bleStatus);
  Serial.println("Bluetooth device active, waiting for connections...");
}

void loop() {
  // put your main code here, to run repeatedly:

}`

Removing one of the attributes, it works, but with the script as above advertise hangs...

solidcloudio, 2017-04-15:
I wouldn't say this is an "Enhancement" but rather a bug..

russmcinnis, 2017-05-02:
We are going to investigate the limits of characteristics when we have V4.2 stabilized.

solidcloudio, 2017-05-02:
Great, thanks.  Let me know if you need help testing

b100bf, 2017-07-03:
I cannot advertise more than 7 characteristics even in separate services. I think there is a memory limitation.
If i'll use 16bit UUIDs instead of 128bit, will it allow more characteristics, or does it only a short representation?