arduino/ArduinoCore-arc32 #449: cannot define size of characteristic

lorenzoromagnoli, 2017-02-22:
I'mt trying to create a characteristic that should store 3bytes. 

I couldn't find a proper documentation but in the source code of the library I found this

```Arduino
/**
     * @brief   Create a characteristic with string value
     *
     * @param   uuid        The UUID of the characteristic
     *
     * @param   properties  The properties of the characteristic
     *
     * @param   value       The string of the characteristic data
     *
     * @return  none
     *
     * @note  The data length is string's size. Can't set a string that is longger
     *          than the this input
     */
    BLECharacteristic(const char* uuid, 
                      unsigned char properties, 
                      const char* value);
```

I tried using the third parameter when I create the characteristic but it does not compile throwing this error

```
bleColor:8: error: no matching function for call to ‘BLECharCharacteristic::BLECharCharacteristic(const char [37], int, int)’
 BLECharCharacteristic color_Characteristic("19B10011-E8F2-537E-4F6C-D104768A1214", BLERead | BLEWrite, 2);
```

Here is my complete code for reference:

```Arduino
#include <CurieBLE.h>

BLEPeripheral blePeripheral; // create peripheral instance
BLEService ledStripService("19B10010-E8F2-537E-4F6C-D104768A1214"); // create service

// create switch characteristic and allow remote device to read and write
BLECharCharacteristic color_Characteristic("19B10011-E8F2-537E-4F6C-D104768A1214", BLERead | BLEWrite, 2);

void setup() {
  Serial.begin(9600);

  // set the local name peripheral advertises
  blePeripheral.setLocalName("ledStrip");
  // set the UUID for the service this peripheral advertises:
  blePeripheral.setAdvertisedServiceUuid(ledStripService.uuid());

  // add service and characteristics
  blePeripheral.addAttribute(ledStripService);
  blePeripheral.addAttribute(color_Characteristic);

  // advertise the service
  blePeripheral.begin();

  Serial.println("Bluetooth device active, waiting for connections...");

    // assign event handlers for connected, disconnected to peripheral
  blePeripheral.setEventHandler(BLEConnected, blePeripheralConnectHandler);
  blePeripheral.setEventHandler(BLEDisconnected, blePeripheralDisconnectHandler);

  
}

void loop() {
  // poll peripheral
  blePeripheral.poll();

  if (color_Characteristic.written()) {
    Serial.println("changed color");  
    Serial.print("dec: ");
    Serial.println(color_Characteristic.value(), DEC);
    Serial.print("hex: ");
    Serial.println(color_Characteristic.value(), DEC);
    Serial.print("bin: ");
    Serial.write(color_Characteristic.value());
  }
 
}

void blePeripheralConnectHandler(BLECentral& central) {
  // central connected event handler
  Serial.print("Connected event, central: ");
  Serial.println(central.address());
}

void blePeripheralDisconnectHandler(BLECentral& central) {
  // central disconnected event handler
  Serial.print("Disconnected event, central: ");
  Serial.println(central.address());
}
```

sandeepmistry, 2017-02-22:
Hey @lorenzoromagnoli,

Instead of:

```
BLECharCharacteristic color_Characteristic("19B10011-E8F2-537E-4F6C-D104768A1214", BLERead | BLEWrite, 2);
```

use the untyped BLECharacteristic:

```
BLECharacteristic color_Characteristic("19B10011-E8F2-537E-4F6C-D104768A1214", BLERead | BLEWrite, 2);

```

The Heart rate monitor example seems like something you might be interested in: https://www.arduino.cc/en/Tutorial/Genuino101CurieBLEHeartRateMonitor

lorenzoromagnoli, 2017-02-23:
Alright, this was my fault!
Now it compiles, but the `color_Characteristic.written()` method never returns true when I send new values. 

Moreover, apparently I cannot print the`color_Characteristic.value()`. 
When I do     
```Arduino
Serial.println(color_Characteristic.value(), DEC); 
```
I get a compiling error saying that

```
call of overloaded ‘println(const __FlashStringHelper*, int)’ is ambiguous
```

What I'm trying to do is to send an array with 3bytes to describe the colour of a led lamp. Unfortunately I couldn't find any tutorial where numeric data are sent from an app to the Arduino, and I'm a bit struggling because of the very minimal documentation provided.

would be happy to share a tutorial once I manage to get this running. 

Thanks for your help @sandeepmistry

lorenzoromagnoli, 2017-02-23:
Did a bit more of research and food a Tutorial doing what I was trying to achieve.

https://create.arduino.cc/projecthub/gov/voronoi101-light-your-lamp-a22a9e?ref=search&ref_id=ble%20lamp&offset=0

I solved adding an event handler for when the characteristic gets written 
```Arduino
color_Characteristic.setEventHandler(BLEWritten, color_CharacteristicWritten);
```
and the following function to parse the received data.

```Arduino
void color_CharacteristicWritten(BLECentral& central, BLECharacteristic& characteristc) {
  const unsigned char* colors = color_Characteristic.value();
  Serial.print(colors[0]);
  Serial.print("-");
  Serial.print(colors[1]);
  Serial.print("-");
  Serial.println(colors[2]);
}
```
I think this could be a very helpful starting point for anyone trying to do something more complicated than switching something On/Off. Would be great to see something like this in the examples.

akash73, 2017-02-23:
thanks @lorenzoromagnoli 
tutorial added.

lorenzoromagnoli, 2017-02-24:
I put together some code to create an app that talks to the 101 via BLE.
[Ionic2 Arduino BLE test](https://github.com/lorenzoromagnoli/Ionic2_arduino_BLE_test)

sandeepmistry, 2017-02-24:
@lorenzoromagnoli nice work!

We should be able to simplify things:

```arduino
void color_CharacteristicWritten(BLECentral& central, BLECharacteristic& characteristc) {
  const unsigned char* colors = color_Characteristic.value();
  Serial.print(colors[0]);
  Serial.print("-");
  Serial.print(colors[1]);
  Serial.print("-");
  Serial.println(colors[2]);
}
```

to:

```arduino
void color_CharacteristicWritten(BLECentral& central, BLECharacteristic& characteristc) {
  Serial.print(color_Characteristic.value[0]);
  Serial.print("-");
  Serial.print(color_Characteristic.value[1]);
  Serial.print("-");
  Serial.println(color_Characteristic.value[2]);
}
```

lorenzoromagnoli, 2017-02-27:
nice

closed 2017-03-06 by kitsunami