arduino/ArduinoCore-arc32 #473: LED not cleared after loading a blank sketch

sandeepmistry, 2017-03-15:
This occurred while testing with the 2.0.1 rc1 JSON.

1) Load sketch to turn LED on:

```arduino
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);
}

void loop() {
}
```

2) LED is on.

3) Load blank sketch:

```arduino
void setup() {
}

void loop() {
}
```

4) LED remains on.

After step 3) I expect the LED to be off as the sketch is empty.

bigdinotech, 2017-03-16:
@sandeepmistry I was able to reproduce but only if extend the delay for when the pin state is HIGH.
Basically its just keeping the state it was at before it was rebooted.
I should have a fix soon.

SidLeung, 2017-03-16:
Tracked by Jira 892.

sandeepmistry, 2017-03-24:
This looks good now with the 2.0.2 RC JSON @SidLeung provided me. Closing.

closed 2017-03-24 by sandeepmistry