arduino/ArduinoCore-arc32 #474: tinyTile Support

mjs513, 2017-03-15:
Have a couple of questions on support for the tinyTile.  
1. Does any have a complete pin mapping for the tinyTile.
2. Are all pins supported by the Arduino 101 library.  I am really interested in the BMU160 ability to have an external sensor attached. 

Thanks
Mike

kitsunami, 2017-03-20:
#1. The tinyTILE product page provides a pin map PDF under their documents section: 
https://www.element14.com/tinyTILE

mjs513, 2017-03-20:
I will stand corrected. You are correct those are the standard pins. What are the labels for the test pads on the front and bottom of the board and the through holes. See discussion between me and @sidleung on the tinytile re adding a magnetometer.

Thank you

SidLeung, 2017-03-20:
Ha-ah!  At least someone remembers me...  Am so flattered.  Unfortunately, whatever I said before was done my evil twin brother.

However, on the more serious side of things...

Looks like you're looking for the PAD layout on the back of tinyTILE.  Did check out the link provided above and it does not include the backside of the board (too boring, wouldn't you say).  @kitsunami , would you be kind enough to share that nice picture we put up on Basecamp tinyTILE?  Or you can give me the permission to put it up here.

BTW, @mjs513 , you still need to visit our good friend, BMI, for the driver to interface with the external magnetometer.  That one, we can't provide it to you without being beaten by an ugly stick.

mjs513, 2017-03-20:
@sidleung and @kitsunmi. The pdf doesn't show the back side of the board. Only shows the edge pins.

@SIDLEUNG Already have the api and the register mapping. :).  Got the fifo working with the curieimu would be nice to have it working a magnetometer :).  Also already have a tinytile in hand.

russmcinnis, 2017-03-28:
Hey Mike, We do use some of the backside signals and have mapped them to pin numbers. I am working with somebody so we can get you something. We have a silkscreen image of the TPB point with numbers also which we use for soldering or pogo pin contacts.

SidLeung, 2017-03-28:
I ping Kit on this issue a week ago.  I know that the backside of the silkscreen is available on BaseCamp.  I was asking her permission to forward the info.  Will ping on her again.

mjs513, 2017-03-28:
Hi @russmcinnis / @SidLeung thanks for the update. I was thinking about the pogo pin contacts as well to get access.

kitsunami, 2017-04-04:
Hi @mjs513 ! 
I've reached out to element14 to see if they can update their product documents to support your request. 
Since this is their product, I recommend using their product forum (https://www.element14.com/tinyTILE) to interact with them directly.

SidLeung, 2017-04-04:
@mjs513 

Element 14 was made aware of the situation that you were looking for the backside silk screen by @kitsunami.  They will update their documentation accordingly.  No timeline given though.

mjs513, 2017-04-04:
@SidLeung and @kitsunami 

Thank you for letting me know that you informed element14 about the documentation issue.  Not really sure what the big deal is that you can not post it if you have it and you can not share.  And this took two weeks to let me know that I should contact element14. 

@SidLeung seriously though thanks for your help in getting sorted out.  By the way where are you from - intel, Arduino etc?  I will go bug them.

Mike

closed 2017-04-04 by mjs513