arduino/ArduinoCore-arc32 #478: CurieBLE: descriptor.read() does not block

sandeepmistry, 2017-03-15:
Similar to #383 but for descriptor.

I would like to remove the `delay(1000);` statements from the peripheral_explorer example. So, expect ` descriptor.read()` to behave like `characteristic.read()` and block.

SidLeung, 2017-03-17:
Tracked by Jira 890.

russmcinnis, 2017-03-21:
We have system tested, but not unit tested yet.

russmcinnis, 2017-03-21:
I was expecting the delay(1000) to be removed from peripheral_explorer.ino so this needs an update.  
// read the descriptor value
  descriptor.read();
  delay(1000);

sandeepmistry, 2017-03-24:
This looks good now with the 2.0.2 RC JSON @SidLeung provided me. Closing.

I'll open a small to remove the `delay(1000);`'s in the `peripheral_explorer.ino` as @russmcinnis mentioned above.

Here's the result from running again bleno's test.js on a Raspberry Pi (I had to mod the sketch to scan with duplicates as mentioned in #370):

```
Found B8:27:EB:1C:39:03 '' ffffffff-ffff-ffff-ffff-fffffffffff0
Found B8:27:EB:1C:39:03 'test' ffffffff-ffff-ffff-ffff-fffffffffff0
Connecting ...
Connected
Discovering attributes ...
Attributes discovered

Device name: 
Service 1800
Service 1801
Service ffffffff-ffff-ffff-ffff-fffffffffff0
	Characteristic ffffffff-ffff-ffff-ffff-fffffffffff1, properties 0x2, value 0x76616C7565
		Descriptor 2901, value 0x75736572206465736372697074696F6E
	Characteristic ffffffff-ffff-ffff-ffff-fffffffffff2, properties 0x2, value 0x64796E616D69632076616C7565
	Characteristic ffffffff-ffff-ffff-ffff-fffffffffff3, properties 0x2, value 0x000102030405060708090A0B0C0D0E0F10111213
	Characteristic ffffffff-ffff-ffff-ffff-fffffffffff4, properties 0x12
	Characteristic ffffffff-ffff-ffff-ffff-fffffffffff5, properties 0x16
	Characteristic ffffffff-ffff-ffff-ffff-fffffffffff6, properties 0x32

Disconnecting ...
Disconnected

```

closed 2017-03-24 by sandeepmistry