arduino/ArduinoCore-arc32 #479: CurieBLE: Discover attributes fails with macOS + bleno test.js app

sandeepmistry, 2017-03-15:
When I run the peripheral explorer example against my macOS 10.12.3 Mac running the test.js example of bleno, the attribute discovery fails. Testing against Raspberry Pi is fine.

```
Connected
Discovering attributes ...
Attribute discovery failed!
```

I've attached a screenshot of the PacketLogger app:

[image]

sandeepmistry, 2017-03-16:
This seems to be a duplicate of #373.

If I remove the last 2 characteristics the bleno test.js app adds, things are fine:

```
Service 1800
Service 1801
Service 180a
	Characteristic 2a29, properties 0x2, value 0x4170706C6520496E63
	Characteristic 2a24, properties 0x2, value 0x4D6163426F6F6B50726F31322C31
Service d0611e78-bbb4-4591-a5f8-487910ae4366
	Characteristic 8667556c-9a37-4c91-84ed-54ee27d90049, properties 0x24
Service 9fa480e0-4967-4542-9390-d343dc5d04ae
	Characteristic af0badb1-5b99-43cd-917a-a77bc549e3cc, properties 0x24
Service ffffffff-ffff-ffff-ffff-fffffffffff0
	Characteristic ffffffff-ffff-ffff-ffff-fffffffffff1, properties 0x2, value 0x76616C7565
		Descriptor 2901, value 0x75736572206465736372697074696F6E
	Characteristic ffffffff-ffff-ffff-ffff-fffffffffff2, properties 0x2, value 0x64796E616D69632076616C7565
	Characteristic ffffffff-ffff-ffff-ffff-fffffffffff3, properties 0x2, value 0x000102030405060708090A0B0C0D0E0F10111213
	Characteristic ffffffff-ffff-ffff-ffff-fffffffffff4, properties 0x12
```

Note the `ffffffff-ffff-ffff-ffff-fffffffffff4` and `ffffffff-ffff-ffff-ffff-fffffffffff5` characteristics from the origin test.js file have been removed.

So, it appears to be a memory limitation.

SidLeung, 2017-04-04:
I believe this issue should be addressed with Deneb.

sandeepmistry, 2017-04-05:
@SidLeung this still happens with Deneb:

```
BLE Central - Peripheral Explorer
Found 3B:5B:E4:E5:C2:BC '' 
Found C8:69:CD:41:84:A2 '' 
Found 56:D4:38:BA:3A:7F '' 
Found AC:BC:32:7C:FA:26 'test' ffffffff-ffff-ffff-ffff-fffffffffff0
Connecting ...
Connected
Discovering attributes ...
Attribute discovery failed!

```

I believe it is memory related and with be resolved in Bootes.

sandeepmistry, 2017-04-11:
This looks good now with the Bootes pre-release JSON:

```
Connected
Discovering attributes ...
Attributes discovered

Device name: 
Service 1800
Service 1801
Service 180a
	Characteristic 2a29, properties 0x2, value 0x4170706C6520496E63
	Characteristic 2a24, properties 0x2, value 0x4D6163426F6F6B50726F31322C31
Service d0611e78-bbb4-4591-a5f8-487910ae4366
	Characteristic 8667556c-9a37-4c91-84ed-54ee27d90049, properties 0x24
Service 9fa480e0-4967-4542-9390-d343dc5d04ae
	Characteristic af0badb1-5b99-43cd-917a-a77bc549e3cc, properties 0x24
Service ffffffff-ffff-ffff-ffff-fffffffffff0
	Characteristic ffffffff-ffff-ffff-ffff-fffffffffff1, properties 0x2, value 0x76616C7565
		Descriptor 2901, value 0x75736572206465736372697074696F6E
	Characteristic ffffffff-ffff-ffff-ffff-fffffffffff2, properties 0x2, value 0x64796E616D69632076616C7565
	Characteristic ffffffff-ffff-ffff-ffff-fffffffffff3, properties 0x2, value 0x000102030405060708090A0B0C0D0E0F10111213
	Characteristic ffffffff-ffff-ffff-ffff-fffffffffff4, properties 0x12
	Characteristic fffffffffffff-ffff-ffff-fffffffffff5, properties 0x16
	Characteristic ffffffff-ffff-ffff-ffff-fffffffffff6, properties 0x32

Disconnecting ...
Disconnected
```

SidLeung, 2017-04-18:
@sandeepmistry Would you please close the issue if the code is satisfactory.

novashah, 2017-04-25:
Closing issue as resolved.

closed 2017-04-25 by novashah