arduino/ArduinoCore-arc32 #485: QUESTION:  nRF51822 2.4GHz Radio Supported?

mjs513, 2017-03-21:
I asked this question on the forum but have no answer so I figured I would add it here. While I was looking at getting some info on the tinyTile I came across that the nRF5188.2 which supposedly is on the Arduino101 also supports not only the BLE protocol stack but also a 2.4GHz protocol stacks.  I believe it uses the Gazelle protocol stack? Is there any plans to turn on this capability?

SidLeung, 2017-03-21:
Hi, @mjs513 ,

Gazell is Nordic's proprietary wireless protocol that can run on the nRF51822 SoC.  There is no plan to support this feature in the Arduino 101 platform.

mjs513, 2017-03-21:
Hi @SidLeung, again. 

Thanks for getting back to me.  It would have been a great addition to the Arduino 101 platform.  haven't seen it anywhere else.  

thanks
Mike

closed 2017-03-21 by mjs513