arduino/ArduinoCore-arc32 #492: Add arduino 101 board into LinuxArm download to be able to upload sketches from raspberry pi

bdavis721, 2017-03-22:


SidLeung, 2017-03-22:
@bdavis721 ,

Would like to know if you were asking a question.

bdavis721, 2017-03-22:
Sorry about the confusing issue. I wanted the ability to add the intel curie boards into the arduino ide for the Linux Arm operating systems because I have an arduino 101 that I want to upload sketches to from a raspberry pi 3 but I cannot download that board in the board manager because the "tool arc-elf32" is not available for my os.

SidLeung, 2017-03-22:
Let me if this is what you want to do:
1.  You want to use your Raspberry Pi as the computing platform that is running a Linux/ARM OS.
2.  Install the Arduino IDE on that platform so that,
3.  You can upload sketches to your Arduino 101 board.

Please note that the Arduino IDE runs on Ubuntu 14.04 OS.  Not sure if that is the OS you have on your Raspberry Pi.  If not, that would be the first issue to overcome.

bdavis721, 2017-03-22:
Yes that is what I want to do. I was able to install the most up to date arduino IDE on the pi. I am running Raspbian Jessie but when I go to install the curie board on the IDE i get an error saying "tool arc-elf32 is not available for my os." I've been checking around and the closest thing to a solution I found was here http://forum.arduino.cc/index.php?topic=428145.0 but I dont know what steps to take to do this.

SidLeung, 2017-03-22:
Ok...  This is something the IDE folks may help.  You may have to manually the tool chain.

bdavis721, 2017-03-22:
I need help manually configuring the tool chain please.

facchinm, 2017-03-23:
@SidLeung if you are going to merge https://github.com/01org/corelibs-arduino101/pull/475, the only tool not yet packaged for ARM Linux will be the toolchain, indeed.
Since there are no instructions anywhere about where the sources are I guessed they were generated by the official ARC toolchain (https://github.com/foss-for-synopsys-dwc-arc-processors/toolchain) but I'm not 100% sure. Could you confirm that?

facchinm, 2017-03-27:
Attached you can find the (unstripped) toolchain for armv7 targeting arc32. It was compiled from the aforementioned repository with `/build-all.sh --no-uclibc  --no-pdf` . It is completely untested so don't use it for anything critical :smile: 
Link: https://mega.nz/#!8xJ2gZqJ!nf8-ZkfEJDNYEMQsbEZUKwL40AixmCh3hdbSTex7iTw

bdavis721, 2017-03-27:
Thanks, I'm working on my senior design project and it would be critical so I think I will wait until they update it with a stable version for all my hobbies.

facchinm, 2017-03-27:
If you could test it and give us a feedback it would surely speedup the process :wink:

chock, 2017-07-31:
@facchinm I'm willing to test your compiled toolchain on raspberry pi.  I've downloaded it and extracted it but I'm not sure where the files should be placed or how my environment should be modified so the IDE can find it.  Would you be able to provide any direction on this?

facchinm, 2017-08-01:
Hi @chock ,
the procedure should be:
* install Arduino101 core via Board manager
* the installer should complain about "no suitable tool found for arc32-gcc-..."
* extract the toolchain wherever you want (for example, `/opt/arc32-gcc`
* modify `~/.arduino15/packages/Intel/hardware/arc32/2.0.2/platform.txt` , line `compiler.path=` to point to the folder you extracted (for example `/opt/arc32-gcc/bin`)
* done :wink:

chock, 2017-08-01:
Here is where I am
installed Arduino 1.8.3 IDE
(removed all existing ~/.arduino and ~/.arduino15 directories from previous experimenting)
with Arduino 101 connected, start IDE
using Board manager attempt to install the "Intel Curie Boards by Intel...." package
at this point I get a java backtrace in the IDE message window

```
 Tool openocd is not available for your operating system.
java.lang.RuntimeException: java.lang.Exception: Tool openocd is not available for your operating system.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Tool openocd is not available for your operating system.
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:86)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more
```

facchinm, 2017-08-02:
Ah, right, the IDE cleans up the folders if the installation fails due to missing tools (I forgot about it...).
So, the updated procedure should be:
* find where your sketchbook is (usually $HOME/Arduino)
* cd in it and create a couple of directories (`mkdir -p hardware/arduino101`)
* enter that directory
* clone the 101 git repo there (`git clone https://github.com/01org/corelibs-arduino101.git`)
* checkout tag 2.0.2 (`git checkout 2.0.2`)
* modify `corelibs-arduino101/platform.txt` line `compiler.path=`
* restart the IDE

You should then get the 101 entry under "Boards". It will only compile, not upload. To do that, you also need `arduino101load` tool from [here](https://github.com/facchinm/arduino101load/releases/download/2.0.1/arduino101load-2.0.1-linux_arm.tar.bz2) and change `tools.arduino101load.cmd.path=` line to pint to the correct folder.

chock, 2017-08-03:
I followed your excellent instructions and I have the board loaded properly but the compiler is choking with errors.  I enabled verbose logs and recompiled and I have include the logs in the attached text file.  There are whitespace warnings at the beginning which I can ignore but the linker is failing due to an apparent architecture mismatch.  

Any ideas?

[compile-errors.txt](https://github.com/01org/corelibs-arduino101/files/1195615/compile-errors.txt)

chock, 2017-08-03:
I noticed in my errors that I posted above that it was linking with the arc700 libraries instead of the quarkse_em ones.  I tried to find what was causing that but could not so I renamed the arc700 directories in the 2 referenced locations and symlinked arc700 to quarkse_em (big hack) and recompiled.  That seems to have resolved that issue.  

Now it fails at the very end during the link:

```
Archiving built core (caching) in: /tmp/arduino_cache_163765/core/core_arduino101_corelibs-arduino101_arduino_101_142ed644d70e1e3fd6b968d702a3efa8.a
Linking everything together...
"/home/pi/bin/arc32-toolchain/bin/arc-elf32-gcc" -nostartfiles -nodefaultlibs -nostdlib -static -Wl,-X -Wl,-N -Wl,-mcpu=quarkse_em -Wl,-marcelf -Wl,--gc-sections "-T/home/pi/Arduino/hardware/arduino101/corelibs-arduino101/variants/arduino_101/linker_scripts/flash.ld" "-Wl,-Map,/tmp/arduino_build_452604/sketch_aug02a.ino.map"  -o "/tmp/arduino_build_452604/sketch_aug02a.ino.elf" "-L/tmp/arduino_build_452604" "-L/home/pi/Arduino/hardware/arduino101/corelibs-arduino101/variants/arduino_101" -Wl,--whole-archive "-larc32drv_arduino101" -Wl,--no-whole-archive -Wl,--start-group "-larc32drv_arduino101" -lnsim -lc -lm -lgcc "/tmp/arduino_build_452604/sketch/sketch_aug02a.ino.cpp.o" "/tmp/arduino_build_452604/core/variant.cpp.o" "/tmp/arduino_build_452604/core/core.a"
/home/pi/bin/arc32-toolchain/bin/../lib/gcc/arc-elf32/6.2.1/../../../../arc-elf32/bin/ld: Error: Linker symbol __SDATA_BEGIN__ not found
/home/pi/bin/arc32-toolchain/bin/../lib/gcc/arc-elf32/6.2.1/../../../../arc-elf32/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino 101.
```

facchinm, 2017-08-03:
Found it: the problem looks due to newlib not being at the correct revision (see https://jira.zephyrproject.org/browse/SDK-17 ). I'll recompile the toolchain without arc700 support, checking out the correct revision and report back.

facchinm, 2017-08-07:
Hi @chock ,
I recompile the toolchain from Intel tag (2016.03 for reference) and there it is:
https://mega.nz/#!J9hmQapQ!OQigBXt7dCobduMZfLGpoYDqBCBtybmhbki7BRouhBk

It should run on any version of RPi since it's compiled for ARMv6 (instead than the previous one which was ARMv7).

chock, 2017-08-07:
Hi @facchinm  -
Thank you for recompiling the toolchain for me to try.  Unfortunately, I'm still getting that link error with this new version.  Please see below when you have a moment:

```
...
Archiving built core (caching) in: /tmp/arduino_cache_189713/core/core_arduino101_corelibs-arduino101_arduino_101_142ed644d70e1e3fd6b968d702a3efa8.a
Linking everything together...
"/home/pi/arduino101/bin/arc-elf32-gcc" -nostartfiles -nodefaultlibs -nostdlib -static -Wl,-X -Wl,-N -Wl,-mcpu=quarkse_em -Wl,-marcelf -Wl,--gc-sections "-T/home/pi/Arduino/hardware/arduino101/corelibs-arduino101/variants/arduino_101/linker_scripts/flash.ld" "-Wl,-Map,/tmp/arduino_build_303474/sketch_aug07a.ino.map"  -o "/tmp/arduino_build_303474/sketch_aug07a.ino.elf" "-L/tmp/arduino_build_303474" "-L/home/pi/Arduino/hardware/arduino101/corelibs-arduino101/variants/arduino_101" -Wl,--whole-archive "-larc32drv_arduino101" -Wl,--no-whole-archive -Wl,--start-group "-larc32drv_arduino101" -lnsim -lc -lm -lgcc "/tmp/arduino_build_303474/sketch/sketch_aug07a.ino.cpp.o" "/tmp/arduino_build_303474/core/variant.cpp.o" "/tmp/arduino_build_303474/core/core.a"
/home/pi/arduino101/bin/../lib/gcc/arc-elf32/4.8.5/../../../../arc-elf32/bin/ld: Error: Linker symbol __SDATA_BEGIN__ not found
/home/pi/arduino101/bin/../lib/gcc/arc-elf32/4.8.5/../../../../arc-elf32/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino 101.
```

facchinm, 2017-08-08:
Right, I've been able to test it and reproduced the same error.
Here's the toolchain (hopefully final) with `-mno-sdata` flag added by default. https://mega.nz/#!N4hmmCjZ!JpNPOJnH7ryJ22zYZeKv6JKikDr0BWzNsxRW3b-puj4

I tested it on a RPi 2 and it works just good :smile: 
Just for reference, I'm pasting here the process:

```
* Add native arc32-elf-gcc path to PATH
* Modify build-elf32.sh line where PATH=$INSTALLPATH/bin:$PATH to PATH=$PATH:$INSTALLPATH/bin
* Add arm toolchain to PATH
* Compile texinfo 6.1 and add to PATH
* Launch ./build-all.sh --no-uclibc --no-pdf --no-multilib --cpu quarkse_em --host arm-linux-gnueabihf --target-cflags "-ffunction-sections -fdata-sections -mno-sdata -fdata-sections -ffunction-sections" --strip
```

chock, 2017-08-09:
@facchinm -
Success!

I had to adjust my platform.txt so it could find dfu-util and I could not get arduino101load to write to /dev/ttyACM0 unless I was root.  It would scroll timeout messages if I tried to upload as a non-root user.  I researched this and tried adjusting permissions on the device but I think it's more complicated than that so I ultimately prefaced the call to arduino101load with "/usr/bin/sudo" in the hardware.txt file.

Thank you for all the help.  Maybe next time I'll be able to build the toolchain myself :)

Chock

FatMax88, 2017-09-11:
Just flagging my interest on this topic.  I'm struggling with the above info but will persist.
FYI, I'm running a Raspberry Pi 3B with Raspbian 9.1 (stretch) and Linux Arm Arduino IDE 1.8.4.

fhackenberger, 2017-12-15:
If you're just interested in flashing on the Raspberry Pi (after building the firmware on your dev machine), here's some code you can follow:

Build your project on your dev machine, following [Arduino CLI](https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc) docs.
```
~/arduino/arduino --pref build.path=./build --board Intel:arc32:arduino_101 --verbose-build --port /dev/ttyACM0 --upload Energymanagement.ino
```
Abort the flashing process, the binary is now in `build/Energymanagement.ino.bin`

Get dfu-util on the Raspberry:
```
ssh pi@192.168.1.1
$ wget https://downloads.arduino.cc/arduino-1.8.5-linuxarm.tar.xz
$ tar xf arduino-1.8.5-linuxarm.tar.xz
$ ./arduino-1.8.5/arduino --install-boards 'arduino:stm32f4' # This is not the board we'are gonna use, but it contains dfu-util
$ mv /home/pi/.arduino15/packages/arduino/tools/dfu-util/0.9.0-arduino1 dfu-util-0.9.0-arduino1
$ rm -Rf arduino-1.8.5-linuxarm.tar.xz arduino-1.8.5
$ # rm -Rf ~/.arduino15 # Be careful with this, if you are using another arduino IDE installation on the raspi
```
for flashing, you only need the `/home/pi/.arduino15/packages/arduino/tools/dfu-util/0.9.0-arduino1` directory after running this.

The arduino package for the Arduino 101 contains the arduino101load tool which
I'll simply build for ARM manually here. It basically just calls dfu-utils.

It's the same tool as here, but for arm: `~/arduino-1.8.5/portable/packages/Intel/tools/arduino101load/2.0.1/arduino101load`
```
cd /tmp
git clone https://github.com/facchinm/arduino101load.git
cd arduino101load/
git checkout 2.0.1
cd arduino101load
export GOPATH=$PWD
export GOBIN=.
go get -d
go build # For local machine
GOARCH=arm GOARM=7 go build # For the raspberry
scp arduino101load pi@192.168.1.1:
```

Now we are ready to flash. I captured the arduino101load command with 'ps aux' while running the upload on my development machine.
```
scp build/Energymanagement.ino.bin pi@192.168.1.1:
# Beware arduino101load needs the absolute paths
ssh pi@192.168.1.1 'sudo ./arduino101load -dfu=/home/pi/dfu-util-0.9.0-arduino1 -bin=/home/pi/Energymanagement.ino.bin -port=/dev/ttyACM0 -q -ble_fw_str="ATP1BLE00R-1631C4439" -ble_fw_pos=169984 -rtos_fw_str="" -rtos_fw_pos=0 -core=2.0.0'
```
This should give you the familiar `Starting download script...`