arduino/ArduinoCore-arc32 #503: BLE chip firmware source code

carlosperate, 2017-03-27:
Hi all,

As far as I understand, there is an internal nRF51822 core providing the BLE capabilities. I'm having a hard time locating the source code for this microcontroller. Is this available in this repository or somewhere else?

SidLeung, 2017-04-04:
Closing this PR with no further questions.

closed 2017-04-04 by SidLeung

carlosperate, 2017-04-04:
Sorry for the late reply, must have missed it from my notifications.
Is there any reason the BLE server source code could not be provided? Even if the soft device is not included.

SidLeung, 2017-04-04:
As you may be aware of the fact that the BLE server code is not a standalone entity developed by Intel.  To open source the entire stack would require substantial resources to sort out the licensing, logistic issues with various modules.  We simply do not have that kind of resources at the moment.

carlosperate, 2017-04-06:
That's a pity, I was hoping it wasn't an arduous task, but I do understand. Thanks for your prompt replies.

SidLeung, 2017-04-06:
If you take a quick peek at the Nordic site with regard to their SDK, it may provide a better understanding of the situation I was talking about.  May be, if you could identify what your needs with the Nordic stack, we might be able to help in other ways.

carlosperate, 2017-04-06:
The licensing of the SDK v13 http://developer.nordicsemi.com/nRF5_SDK/nRF5_SDK_v13.x.x/doc/13.0.0/licenses.html seems to be a lot more simpler, but being relatively new I assume the server code will probably be using an older version.
I believe the newer version of the 101 core will support central mode, so I assume it must be using the s130 or s132 soft device? If that's the case maybe this new version could move to the S132 from v13?

The one reason I was asking about this is to consider if it was possible able to access the radio hardware directly to implement very simple protocols and to possibly get a micro:bit talking with an arduino 101 over this interface:
https://lancaster-university.github.io/microbit-docs/ubit/radio/
https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitRadio.cpp

carlosperate, 2017-04-06:
Nevermind about the v13, just noticed it does not support nRF51.

SidLeung, 2017-04-06:
@carlosperate 

Please note that we made use of the BLE stack, a binary image, from Nordic to implement that BLE protocol.  There is no access to the underneath SoC h/w.  You would have to explore other sources to see if that is feasible.

carlosperate, 2017-04-06:
That's why I was hoping to see the code, I wasn't quite sure how an internal core like this would be configured.
In normal nRF51 micros the radio registers are located at `0x40001000` and then just needed to access the registers without the need for any soft device present ([nrf51.h#L297-L366](https://github.com/ARMmbed/mbed-os/blob/729ef153076ea9db4a9c9f81e90076960e035213/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/sdk/source/device/nrf51.h#L297-L366) )

SidLeung, 2017-04-06:
@carlosperate 
The BLE server code employs the API's offered by the Nordic Soft Device.  To access the underneath h/w directly would have made our server code not portable (reusable) any more, that's deemed undesirable.  How does the Soft Device accesses the underneath h/w is beyond our scope of visibility.

carlosperate, 2017-04-06:
Fair enough, thanks @SidLeung !