arduino/ArduinoCore-arc32 #508: Define functional requirements for BLE.scan() to process duplicated Peripherals.

SidLeung, 2017-03-28:
The BLE.scan API is (in v 2.0.2) capable of filtering out duplicated detected Peripherals.  The fact that Advertisement data of a Peripheral may be presented in one or multiple packets (Adv Data and Scan Response Data).  Thus, this ticket is to clarify the requirements of the scan() API on how to process the Adv info from different kinds of Peripheral.

SidLeung, 2017-03-28:
Capturing a discussion on this issue here,

https://github.com/01org/corelibs-arduino101/pull/500#discussion_r108316928

SidLeung, 2017-03-28:
@sandeepmistry @sgbihu , please present your suggestions/requests for change here.

sgbihu, 2017-03-29:
The scan request will send out after the central see the advertising. So the scan response will be got in an expected time. The implementation can add a delay to return new device. I have implement a solution here https://github.com/01org/corelibs-arduino101/pull/502

novashah, 2017-04-25:
Fixed and Merged.

closed 2017-04-25 by novashah