arduino/ArduinoCore-arc32 #509: Latest Code Fails to compile

solidcloudio, 2017-03-29:
Regardless of the script I try to compile I get:

exec: "C:\\Users\\xxx\\AppData\\Local\\Arduino15\\packages\\Intel\\tools\\arduino101load\\1.6.9+1.28/arduino101load": file does not exist
Error compiling for board Arduino/Genuino 101.

This folder does exist however.  Any ideas?

SidLeung, 2017-03-29:
Would you happen to download the JSON here,

https://basecamp.com/2546761/projects/7617309/messages/67770509#comment_515044686

solidcloudio, 2017-03-29:
Huh?  Basecamp link?

sandeepmistry, 2017-03-29:
You will need to update the tools required by installing the pre-release Arduino 101 core with this JSON in the IDE preferences and installing "Intel test" in the board manager:

https://github.com/01org/corelibs-arduino101/releases/download/2.0.1-RC2.1/package_public_2.0.1-rc2.1_index.json

solidcloudio, 2017-04-01:
Hmmm, that doesnt exist.  I still can't seem to get the latest stuff to compile

facchinm, 2017-04-05:
@solidcloudio the problem is 99% related with an antivirus which recognizes `arduino101load` as a threat. Reinstall the latest core and whitelist the file and you should be ok

SidLeung, 2017-04-12:
Closing this issue with no further inputs.

closed 2017-04-12 by SidLeung

solidcloudio, 2017-04-15:
2.0.2 is working other than the BLE more than 6-7 attributes issue