arduino/ArduinoCore-arc32 #511: Curie can not compile function itoa ()

cnlg, 2017-03-29:
**Sample code**

`#include <stdlib.h> 　　  
#include <stdio.h>

int number = -10;
char *string = "test";

void setup() {
  Serial.begin(115200);
  while(!Serial);
  Serial.println("*************Itoa test**************");
  itoa(number,string,10);

}

void loop() {
  Serial.println("*************Result**************");
  Serial.println(string);
  while(1);

}`

**Run the above procedure to get the results are as follows**

*************Itoa test**************
*************Result**************
test

**how to solve this problem？**

cnlg, 2017-03-29:
Curie can not compile function itoa ()

eriknyquist, 2017-03-29:
`itoa()` compiles just fine. The actual warning from the compiler is

```
    C:\Users\enyquist\AppData\Local\Temp\arduino_modified_sketch_123534\sketch_mar29a.ino:5:16: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

    char *string = "test";
```

In C++, string literals are always const, which the warning above hints at. Try using an array instead:

```
#include <stdlib.h>
#include <stdio.h>

int number = -10;
char string[5]; // <---------------------

void setup() {
  Serial.begin(115200);
  while(!Serial);

  Serial.println("Itoa test*");
  itoa(number, string, 10);
}

void loop() {
  Serial.println("Result*");
  Serial.println(string);
  while(1);
}
```

cnlg, 2017-03-30:
thanks ! @eriknyquist

closed 2017-03-30 by eriknyquist