arduino/ArduinoCore-arc32 #517: CurieBLE: characteristic.writeInt(...) reports success when peripheral responds with error

sandeepmistry, 2017-03-29:
Similar to #516, but for write requests. We should also see if the same applies to `characteristic.subscribe()`.

Testing with the 2.0.1RC2.1 JSON:

Arduino sketch:

```arduino
#include <CurieBLE.h>

void setup() {
  Serial.begin(9600);

  while (!Serial);

  Serial.println("BLE Central - Read + Write test");

  // initialize the BLE hardware
  BLE.begin();

  // start scanning for peripheral
  BLE.scanForName("rwfail");
}

void loop() {
 // check if a peripheral has been discovered
  BLEDevice peripheral = BLE.available();

  if (peripheral) {
    // discovered a peripheral, print out address, local name, and advertised service
    Serial.print("Found ");
    Serial.println(peripheral.address());

    BLE.stopScan();

    Serial.println("Connecting ...");
    if (peripheral.connect()) {
      Serial.println("Connected");
    } else {
      Serial.println("Failed to connect!");
      while(1);
    }
    
    // discover peripheral attributes
    Serial.println("Discovering attributes ...");
    if (peripheral.discoverAttributes()) {
      Serial.println("Attributes discovered");
    } else {
      Serial.println("Attribute discovery failed.");
      peripheral.disconnect();
      while(1);
    }

    BLECharacteristic c = peripheral.characteristic("fa12");

    while (peripheral.connected()) {
      Serial.println("reading ...");
      if (c.read()) {
        // should not happen
        Serial.println("read success");
      } else {
        // expected
        Serial.println("read fail");
      }

      Serial.println("writing ...");
      if (c.writeInt(42)) {
        // should not happen
        Serial.println("write success");
      } else {
        // expected
        Serial.println("write fail");
      }

      delay(1000);
    }

    BLE.scan();
  }
}
```

Node.js bleno test app:

```js
var bleno = require('bleno');

var BlenoPrimaryService = bleno.PrimaryService;
var BlenoCharacteristic = bleno.Characteristic;

console.log('bleno - read + write fail');

bleno.on('stateChange', function(state) {
  console.log('on -> stateChange: ' + state);

  if (state === 'poweredOn') {
    bleno.startAdvertising('rwfail', ['fa11']);
  } else {
    bleno.stopAdvertising();
  }
});

bleno.on('advertisingStart', function(error) {
  console.log('on -> advertisingStart: ' + (error ? 'error ' + error : 'success'));

  if (!error) {
    bleno.setServices([
      new BlenoPrimaryService({
        uuid: 'fa11',
        characteristics: [
          new BlenoCharacteristic({
            uuid: 'fa12',
            properties: ['read', 'write'],
            onReadRequest: function(offset, callback) {
              console.log('onReadRequest');
              callback(BlenoCharacteristic.RESULT_UNLIKELY_ERROR);
            },
            onWriteRequest: function(data, offset, withoutResponse, callback) {
              console.log('onWriteRequest');
              callback(BlenoCharacteristic.RESULT_UNLIKELY_ERROR);
            }
          })
        ]
      })
    ]);
  }
});

```

`characteristic.writeInt(42)` is return true even though the GATT write request fails. I expect it to return false to indicate failure.

Packet Logger trace:

[image]

noelpaz, 2017-03-29:
Issue confirmed:

[issue517.zip](https://github.com/01org/corelibs-arduino101/files/880666/issue517.zip)

[image]

novashah, 2017-04-25:
@sandeepmistry Please review and close.

sandeepmistry, 2017-05-02:
Hi @novashah,

I believe this is still an issue, since PR #518 (or an equivalent) has not been merged in.

mxwang101, 2017-08-15:
@noelpaz :+1: Cloud you please tell me what kinds of tools you used for to Packet Logger trace? I have same problem, but no tools to trace it. Thanks