arduino/ArduinoCore-arc32 #526: Upload Problem: touching COM with 1200 baud rate cannot reset all the board on Windows 7.

jimaobian, 2017-04-05:
Hi,
Our customer, a teacher, bought a bunch of Arduino 101 from us. However About half of Arduino 101 fail to upload the sketch through the computer in his school (Windows 7), while another half can work properly. By the way all the computers can upload sketches to Arduino Uno.

The other interesting thing is his own laptop with windows 10 can always upload successfully. 

After remote assistance with Teamviewer,  we found the issue that touching serial port with 1200 cannot reset the Arduino 101,  which prevents the upload process. If we reset the board manually when uploading, it works like charm.

We know refresh installing Windows can deal with the problem. However in school there are hundreds of PC. The deployment cost is too high for them. Could you try to research into this problem?  We are delighted to help you dealing with this problem. Thank you

OS: Windows 7 x64
What we have done:
* Uninstall and reinstall the Serial port driver and DFU driver.
* Replace the driver files like usbser.sys, mdmcpq.inf, mdmcpq.PNF in Windows.
* Use CoolTerm to connect the COM with 1200 baud.

Best regards
Angelo

bigdinotech, 2017-04-17:
@jimaobian First thing I would suggest is to try out the latest core-libs release.

My next suggestion is to make sure they are picking the correct serial port on the Arduino IDE, I myself have made this mistake several times.

Has your customer tried a board that was not working on one computer on a computer that was working? Also the other way around a known working board on a non working computer.

Since baud rate reset does not seem to work, while upload with hard reset works, that tells me that DFU is working properly while the serial driver may not.
When the board is connected does it it enumerate as a serial port on the device manager?
Also if they upload  sketch that outputs something to the serial port does it work?

jimaobian, 2017-04-18:
@bigdinotech 
Hi, the board package is 1.0.7 at that time. Not yet test 2.0.2.

The serial port is rightly selected. I even manually change the COM port number at device manager, also use CoolTerm to connect it with 1200 baud.

The board did not work on one computer(all Windows 7 PCs in his school), while on a computer(window 10 PC of his own) it worked.

It does enumerate as a serial port and the driver is well installed via reinstalling the board package.

I didn't try uploading sketch that outputs something to the serial port yet.

**BTW: The "TimedWakeup" example of "Arduino101Power" can reproduce this problem. Only Master Reset can solve it.**

BallscrewBob, 2017-04-18:
I can confirm this issue also on Arduino CREATE under win 7 x64 pro. ( Which should be using the latest cores AFAIK )
As per OP the port enumerates correctly.
Using 1.0.6 package in the desktop IDE (1.8.2) and same issue.
Did the 2.0 update and that went through just fine with no reported issues.
Problem persists.

Usually throw "AnalogReadSerial" or std "blink" examples at boards to double check they are playing nice.
Not in this case.

facchinm, 2017-04-18:
Hi everyone, let's try to reorder the information a bit:
- some windows 7 PCs are unable to correctly perform the "1200bps touch" on the serial port.
- by moving the same 101 boards on another PC, the problem disappears.

In fact, the USB-to-serial communication is handled by the x86 firmware, which takes care of issuing a reboot once it receives this "command".  This is why "Timed Wakeup" doesn't work, because the x86 chip is sleeping and can't receive data.

Neither the boards nor the firmware are "broken", there is something fragile in the whole procedure that becomes clear with the Win7 PCs. Are they different in any way form the "working" ones? USB controllers (2.0 vs 3.0) or something else? This would greatly help isolating the issue.

Being able to reproduce the problem is even more important since we are preparing an update for the x86 FW code and this could correct the issue, so we need better information on the PCs setup. 1.0.6 - 1.0.7 and 2.0.2 all share the same fw code so they also share this behaviour.

BallscrewBob, 2017-04-18:
NO usb 3.0 as I am aware of that issue.
I dont have my server available for tests right now (Win 8 to 10 available via that) and only have a dual boot XP / Vista test box that is live. with IDE's 1.6.5 r5 / 1.6.12 / 1.8.1 / 1.8.2.

Just as a side test I grabbed the hourly IDE and ran that.
Apart from having to use master reset each time as it instructs then that appears to be working fine.

Not sure if there is work ongoing in CREATE (prod but have access to dev too) as am now not seeing the touch error in the same way.
`
arduino-builder/arduino-builder -compile -core-api-version 10611 -build-path /tmp/347849427 -hardware arduino-builder/hardware -hardware arduino-builder/packages/cores -tools arduino-builder/tools -tools arduino-builder/packages/tools -built-in-libraries arduino-builder/latest -libraries /tmp/202651701/pinned -libraries /tmp/202651701/custom -fqbn Intel:arc32:arduino_101 -build-cache /tmp -verbose=true /tmp/202651701/101_blink_curie_timer
Using board 'arduino_101' from platform in folder: /home/admin/builder/arduino-builder/packages/cores/Intel/arc32
Using core 'arduino' from platform in folder: /home/admin/builder/arduino-builder/packages/cores/Intel/arc32
Detecting libraries used...
"/home/admin/builder/arduino-builder/packages/tools/arc-elf32/1.6.9+1.0.1/bin/arc-elf32-g++" -c -mcpu=quarkse_em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char -fno-rtti -fno-exceptions -fcheck-new -D__ARDUINO_ARC__ -std=c++11 -DCONFIG_BLUETOOTH_PERIPHERAL -DCONFIG_BLUETOOTH_CENTRAL -DCONFIG_BLUETOOTH_GATT_CLIENT -w -x c++ -E -CC -DF_CPU=32000000L -DARDUINO=10611 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/system/libarc32_arduino101/common" "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/system/libarc32_arduino101/drivers" "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/system/libarc32_arduino101/bootcode" "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/system/libarc32_arduino101/framework/include" "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/variants/arduino_101" "/tmp/347849427/sketch/101_blink_curie_timer.ino.cpp" -o "/dev/null"
"/home/admin/builder/arduino-builder/packages/tools/arc-elf32/1.6.9+1.0.1/bin/arc-elf32-g++" -c -mcpu=quarkse_em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char -fno-rtti -fno-exceptions -fcheck-new -D__ARDUINO_ARC__ -std=c++11 -DCONFIG_BLUETOOTH_PERIPHERAL -DCONFIG_BLUETOOTH_CENTRAL -DCONFIG_BLUETOOTH_GATT_CLIENT -w -x c++ -E -CC -DF_CPU=32000000L -DARDUINO=10611 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/system/libarc32_arduino101/common" "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/system/libarc32_arduino101/drivers" "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/system/libarc32_arduino101/bootcode" "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/system/libarc32_arduino101/framework/include" "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/variants/arduino_101" "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/libraries/CurieTimerOne/src" "/tmp/347849427/sketch/101_blink_curie_timer.ino.cpp" -o "/dev/null"
"/home/admin/builder/arduino-builder/packages/tools/arc-elf32/1.6.9+1.0.1/bin/arc-elf32-g++" -c -mcpu=quarkse_em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char -fno-rtti -fno-exceptions -fcheck-new -D__ARDUINO_ARC__ -std=c++11 -DCONFIG_BLUETOOTH_PERIPHERAL -DCONFIG_BLUETOOTH_CENTRAL -DCONFIG_BLUETOOTH_GATT_CLIENT -w -x c++ -E -CC -DF_CPU=32000000L -DARDUINO=10611 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/system/libarc32_arduino101/common" "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/system/libarc32_arduino101/drivers" "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/system/libarc32_arduino101/bootcode" "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/system/libarc32_arduino101/framework/include" "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/Intel/arc32/variants/arduino_101 [... cut ...]
"/home/admin/builder/arduino-builder/packages/tools/arc-elf32/1.6.9+1.0.1/bin/arc-elf32-objcopy" -S -O ihex -R .note -R .comment -R COMMON -R .eh_frame "/tmp/347849427/101_blink_curie_timer.ino.elf" "/tmp/347849427/101_blink_curie_timer.ino.hex"
Using library CurieTimerOne at version 1.0 in folder: /home/admin/builder/arduino-builder/packages/cores/Intel/arc32/libraries/CurieTimerOne
Sketch uses 49948 bytes (32%) of program storage space. Maximum is 155648 bytes.
Restarting in bootloader mode
Get port list before reset
[COM1 COM16 COM26 COM24 COM25 COM6 COM5 COM4 COM15]
Touching port COM15 at 1200bps
Reset before upload: 1200bps Touch: Open port COM15: Invalid serial port

`

BallscrewBob, 2017-04-18:
Brought my server back online (Server 2012) so windows 8/0 /8.1 base.
No issues with CREATE.
No issues with IDE 1.8.2.

From that I can only surmise that it is a windows 7 specific issue.

jimaobian, 2017-04-19:
Hi, @facchinm, Thanks for the summery and explanation of the sleep mode. It is more clear now.
The most interesting thing is 
* The windows 7 PCs, which are unable to correctly perform the "1200bps touch" on the serial port, can correctly perform it using another Arduino 101.

I agree with the way you target the issue. Logically there are two types of issues that can cause this problem. 
* Windows is successfully connect the serial port with 1200 baud but x86 on Curie fails to receive it or some other works may block it.
* There are some problem or some old cache in the CDC driver, which prevents the Windows to send the negotiation command to x86 on Curie. At this situation, x86 on Curie cannot even get the signal.

Is there any easy way to differ from the two types above?

Thanks in advance.

limitedAtonement, 2020-08-01:
Maybe we can close this one unless @jimaobian has anything to add?

BallscrewBob, 2020-08-02:
All done here with it and not seen issue put through forum either so probably dealt with a looooong time ago.