arduino/ArduinoCore-arc32 #531: BLESubscribed, BLEUnsubscribed Event Handlers are not called

vshymanskyy, 2017-04-12:
For example, in following scenario (Nordic UART Service implementation):
```
#include <CurieBLE.h>

BLEService        NUS    ("713D0000-503E-4C75-BA94-3148F18D941E");
BLECharacteristic rxChar ("713D0003-503E-4C75-BA94-3148F18D941E", BLEWrite | BLEWriteWithoutResponse, BLE_MAX_ATTR_DATA_LEN);
BLECharacteristic txChar ("713D0002-503E-4C75-BA94-3148F18D941E", BLERead  | BLENotify,               BLE_MAX_ATTR_DATA_LEN);

void setup() {
  Serial.begin(9600);

  // begin initialization
  BLE.begin();

  // set the local name peripheral advertises
  BLE.setLocalName("NUS test");

  BLE.setAdvertisedService(NUS);

  // add the characteristic to the service
  NUS.addCharacteristic(rxChar);
  NUS.addCharacteristic(txChar);

  // add service
  BLE.addService(NUS);

  // assign event handlers for connected, disconnected to peripheral
  BLE.setEventHandler(BLEConnected, blePeripheralConnectHandler);
  BLE.setEventHandler(BLEDisconnected, blePeripheralDisconnectHandler);

  // assign event handlers for characteristics
  rxChar.setEventHandler(BLEWritten,      rxCharWritten);
  txChar.setEventHandler(BLESubscribed,   txCharSubscribed);
  txChar.setEventHandler(BLEUnsubscribed, txCharUnsubscribed);

  // set an initial value for characteristics
  unsigned char empty[0] = {};
  rxChar.setValue(empty, 0);
  txChar.setValue(empty, 0);

  // start advertising
  BLE.advertise();

  Serial.println("Bluetooth device active, waiting for connections...");
}

void loop() {
  // poll for BLE events
  BLE.poll();
}

void blePeripheralConnectHandler(BLEDevice central) {
  // central connected event handler
  Serial.print("Connected event, central: ");
  Serial.println(central.address());
}

void blePeripheralDisconnectHandler(BLEDevice central) {
  // central disconnected event handler
  Serial.print("Disconnected event, central: ");
  Serial.println(central.address());
}

void txCharSubscribed(BLEDevice central, BLECharacteristic ch) {
    Serial.println("Subscribed");
}

void txCharUnsubscribed(BLEDevice central, BLECharacteristic ch) {
    Serial.println("Unsubscribed");
}

void rxCharWritten(BLEDevice central, BLECharacteristic ch) {
    const uint8_t* data = ch.value();
    uint32_t len = ch.valueLength();
    Serial.print("Got data:");
    Serial.write(data, len);
}

```

sandeepmistry, 2017-04-12:
@vshymanskyy thanks for the excellent issue report! I've confirmed the issue on my board using the sketch you provided.

@SidLeung can something at Intel look into this please?

russmcinnis, 2017-04-12:
I reproduced the results using another 101 as the central and the sketch in this issue is not seeing the subscribe/unsubscribe events.

russmcinnis, 2017-04-12:
tracked by jira 913

russmcinnis, 2017-05-02:
The event handlers are being called. Jira 913 has been updated and resolved. This is not released yet.

russmcinnis, 2017-05-10:
Will re-test when merged for Elnath.

russmcinnis, 2017-05-17:
testing other BLE issues, this will be re-tested when we get something final for Elnath.

d1runberg, 2017-05-22:
Any news on this?

SidLeung, 2017-05-22:
Issue duplicated, failure determined, correction implemented, and it is in the pipeline for official release.

russmcinnis, 2017-05-30:
Will retest for the official release.

russmcinnis, 2017-06-06:
will test when bootes/elnath released.

russmcinnis, 2017-10-24:
Maybe Dino can provide info
@bigdinotech

vshymanskyy, 2017-10-30:
@bigdinotech hey are there any news?

vshymanskyy, 2017-11-10:
ping ping ping ))

vshymanskyy, 2017-12-21:
Guys, what's the plan?