arduino/ArduinoCore-arc32 #536: FOTA support on Arduino 101?

jimaobian, 2017-04-25:
Hi, 
Is there any plan or roadmap that Arduino 101 will support FOTA? It seems that the hardware of the Arduino 101 can support it and it will be a delicious feature.

Thanks

SidLeung, 2017-04-25:
Hello,

There is no plan to support Firmware Over The Air at this point.  That said, however, we would be providing the basic tools to build such a feature.  Namely, we are planning to release a version of the BLE firmware that has high bandwidth performance so that large data object can be transferred at the maximum speed BLE can offer.

jimaobian, 2017-04-25:
Thanks for your fast reply!

It's indeed a pity. I would highly recommend to develop it. It is perhaps the last step that Arduino 101 can be adopted into real products. 

I have to say the BLE library is great. It is far more easy than developing with nrf51822. The PME library is awesome, especially the document of "DrawingInTheAir" example. After reading it, there is no need to read the long and boring technical document.  The missing low power library can also be found here. We like the Arduino ecosystem. However when we would like to make a real product, we have to re-develop them because of the missing of the FOTA.

If I want to develop it by myself, is there any reference or tutorial, which I can get start with?

Thanks a lot.

novashah, 2017-05-02:
This is not on our roadmap yet.  BLE needs to stablize and then be optimized.

closed 2017-06-01 by novashah