arduino/ArduinoCore-arc32 #545: disconnect after sending Notifications

hgflas, 2017-05-11:
Hello everybody,

I'm just trying to send messages from the Arduino to an Android client via BLE. For this I use the library CurieBLE version 2.0.

To send as many packages as possible, I use notifications. The goal is to generate as much throughput as possible. However, the Arduino always disconnected after a few notifications, if the connection interval is greater than 12. What can be the reason and which class is responsible for the dissconnect?

The code:

`BLEFloatCharacteristic valueCharacteristic("19B10000-E8F2-537E-4F6C-D104768A1216", BLERead | BLENotify);`

```
while (central.connected()) {
         while ((endtime - starttime) <= 60000) // do this loop for up to 1min
         {
          
            float sensorValue  = analogRead(A0); 
            boolean succes = valueCharacteristic.setValue(sensorValue);  //send Value per Notification
         

            endtime = millis();
          }
}

```

sandeepmistry, 2017-05-11:
Hi @hgflas,

Could you please provide a full sketch for this issue. Thanks.

hgflas, 2017-05-11:
```
#include <CurieBLE.h>

BLEService streamService("19B10000-E8F2-537E-4F6C-D104768A1213");
BLEFloatCharacteristic valueCharacteristic("19B10000-E8F2-537E-4F6C-D104768A1216", BLERead | BLENotify);  // remote clients will be able to get notifications if this characteristic

BLEIntCharacteristic switchCharacteristic("19B10001-E8F2-537E-4F6C-D104768A1214", BLERead | BLEWrite);

long previousMillis = 0;
const int analogPin = 0;

unsigned long ulMillisLast = millis();
unsigned long ulMillisInterval = 10;//ms
int counter = 0;

long starttime;
long endtime;

byte merge[20]; //Buffer
int beginIndex = 0;

void setup() {
  Serial.begin(9600);    // initialize serial communication

  // begin initialization
  BLE.begin();

  BLE.setLocalName("Stream Sensor");
  BLE.setDeviceName("Stream Sensor");
  BLE.setAdvertisedService(streamService.uuid());  // add the service UUID

  streamService.addCharacteristic(valueCharacteristic);
  streamService.addCharacteristic(switchCharacteristic);
  BLE.addService(streamService); 

  //BLE.setEventHandler(BLEConParamUpdate, bleCentralConnectionParameterUpdateHandler);
  BLE.setAdvertisingInterval(500);
  // start advertising
  BLE.advertise();

  Serial.println("Bluetooth device active, waiting for connections...");
}

void loop() {
  // listen for BLE peripherals to connect:
  BLEDevice central = BLE.central();

  // if a central is connected to peripheral:
  if (central) {

    Serial.println("Connected ");

    while (central.connected()) {
      if (switchCharacteristic.written()) {
        Serial.println("Characteristic is written ");
        if (switchCharacteristic.value()) {   // any value other than 0

          Serial.println("Übertragen beginnen");
          setConnIntervall(central,150, 150, 0, 20000); 
          delay(5000);

          starttime = millis();
          endtime = starttime;
          while ((endtime - starttime) <= 60000) // do this loop for up to 1min
          {
          
            float sensorValue  = analogRead(A0); //float = 4 byte
            boolean succes = valueCharacteristic.setValue(sensorValue); 
            valueCharacteristic.valueUpdated();
            Serial.print(counter);Serial.print("  Wert:  ");Serial.print(sensorValue);Serial.println(succes);
             
            counter++;
            endtime = millis();
          }
     
          //switchCharacteristic.setValue(0x00);
          setConnIntervall(central, 100, 150, 4, 20000);
          Serial.print("Ende");
        } else {
          Serial.print("Übertragung abbrechen");
        }
      }

    }

    Serial.print("Disconnected from central: ");
    Serial.println(central.address());
  }else{
  
    }
}

void bleCentralConnectionParameterUpdateHandler(BLEDevice central) {
  Serial.println("Updated connection parameter");
  Serial.println("-----------------------");

  // print address
  Serial.print("Address: ");
  Serial.println(central.address());

  Serial.print("Interval: ");
  Serial.println(central.getConnectionInterval());
  Serial.print("Timeout: ");
  Serial.println(central.getConnectionTimeout());
  Serial.print("Latency: ");
  Serial.println(central.getConnectionLatency());

  //Serial1.println();
}

void setConnIntervall(BLEDevice central, int conMin, int conMax, int latency, int timeout) {
  central.setConnectionInterval(conMin, conMax, latency, timeout); //12 = minimum = 11.25ms
  //central.setConnectionInterval(17, 17);
  int intervall = central.getConnectionInterval();
  Serial.print("Intervall "); Serial.println(intervall);
}

```

novashah, 2017-06-01:
@sandeepmistry Please reply, thanks!

Timothdev, 2018-11-04:
Hi @hgflas , 
Did you solve it?