arduino/ArduinoCore-arc32 #550: "package_public-2.1.0-rc1_index.json" does not include ble_core.bin version 2.1.0

jimaobian, 2017-05-18:
When burning bootloader, the following error occurs:
```
dfu-util: Could not open file /Users/Angelo/Library/Arduino15/packages/Intel/tools/arduino101load/2.0.0/firmwares/2.1.0/ble_core.bin for reading: No such file or directory
```

yashaswini-hanji, 2017-05-18:
Please note that, this is a pre-release for our internal testing. Respective ble_core.bin and quark.bin will be manually placed in Arduino15/packages/Intel/tools/arduino101load/2.0.0/firmwares/2.1.0 during test. Kindly, do not use this tag till we have a official release. Currently we are testing the new features and will have a official release in few days. Thanks!

eriknyquist, 2017-05-18:
Or at least, if you do play with it, save the bug reports for the official release :)

yashaswini-hanji, 2017-05-18:
This is not a bug :).. It is looking for ble_core.bin in 2.1.0 directory. Since it is not official release we have not updated aruino101load tool yet.

jimaobian, 2017-05-19:
@eriknyquist 👌, got it.

closed 2017-05-19 by jimaobian